openmv/openmv #42: The question of openmv version

torobot, 2015-02-26:
Access https://github.com/iabdalkader/openmv/tree/master/eagle I found openmv1, openmv2, openmv3 three versions.
1.openmv1 using STM32F407 chips.
2.openmv2 using STM32F429 chip and use SDRAM.
3.openmv3 using STM32F407 chips.
4. In your kickstarter website, the hardware is used STM32F429, do not use SDRAM.

But you have only one source code, I see there are a makefile choice (49 lines to 53 lines), there openmv1 parameter is DSTM32F407xx, default is DSTM32F429xx.

My question is:
1.Above four hardware versions can use your source code (https://github.com/iabdalkader/openmv)?
2.In openmv2 ,SDRAM without any use?(makefile no choice SDRAM)

I'm sorry ,my English is so bad.

iabdalkader, 2015-02-26:
Hi, to confirm we will use openmv3 and an STM32F429/180MHz (they are pin compatible) OMV3-code is Not in the repo yet, I've been working locally, until I figure out what to do with all the mess of different versions.

With some hacking you can get the code to work on OMV3 (if you build one)

SDRAM code will build when BOARD=OPENMV2 is passed on the command line.

torobot, 2015-02-27:
If I want to participate OMV3 project, how can I get the code? (Or so far has not been open source)

iabdalkader, 2015-02-27:
It's basically the same code, with some changes to account for the different pin out/connections and STM32F429, I will add those changes very soon, to this repo, will probably move all other boards to a different repo.

torobot, 2015-02-27:
Well done, thank you, very much looking forward as soon as possible.

closed 2015-02-27 by torobot