openmv/openmv #43: Failed to set custom baud rate: 12000000

superted6851, 2015-02-27:
When connecting to the IDE this error "Failed to set custom baud rate: 12000000" pops up on Ubuntu 14.04 64bit, Ubuntu 13.10 32bit and windows 7 32 bit. All these machines use pyserial==2.7. If you reduce the baud rate to 4000000 the error does not occur, but in linux you still can not connect to the IDE as the baud does not equal the debug value set in micropython/stmhal/usbd_cdc_interface.c, which leads to a seperate error pyserial error. This does not occur in windows 7.

shimniok, 2015-02-27:
I'll bet that's what's going on with the Mac -- it cannot connect unless a
standard baud rate is passed to PySerial. But I didn't edit the
usbd_cdc_interface.c ... when I get a chance I'll try that. Thanks.

iabdalkader, 2015-02-27:
I have pyserial 2.7 too but I've been testing on Arch, will try again on Ubuntu, probably doesn't matter anyway, unless it's slower.

iabdalkader, 2015-07-04:
I can't reproduce this (at least not with the latest FW/IDE), I just tested that custom baudrate  on Arch and Windows 7 32-bit.

PS: Python version won't matter on Windows because I'll package the whole thing as a single binary.

iabdalkader, 2015-07-07:
Okay found the problem, you need to pass 2.7 to pyserial, version 2.7 allows custom baudrates:

`sudo pip install pyserial==2.7`

On Windows and Arch we're using 2.7 by default I think

This patch was merged in 2.7, see: 
http://digital.ni.com/public.nsf/allkb/D37754FFA24F7C3F86256706005B9BE7

closed 2015-07-07 by iabdalkader

iabdalkader, 2015-07-08:
This doesn't seem to affect the FPS at all, so I'm using the highest standard baudrate (921600) and the custom one will still work and can be set from the IDE.