openmv/openmv #46: Generated header files

waldo3000, 2015-05-16:
When the makefile for OpenMV is run, it runs the MicroPython makefile which generates some customized headers in the genhdr directory.  I'm having some problems getting this process to work correctly.

Is it possible to put the files in the genhdr directory up on GitHub so I can use those rather than re-generating them myself?

Thanks.

iabdalkader, 2015-05-16:
No those files are auto-generated, what problems are you having ?

iabdalkader, 2015-05-16:
If you were having problems cloning the repo try again, I just reverted the MicroPython submodule to https shouldn't ask for authentication
```
git clone --recursive https://github.com/iabdalkader/openmv.git
```

waldo3000, 2015-05-17:
Initially I had problems getting my preferred IDE (Rowley Crossworks) to run the py files, but I think I can run them manually outside of the makefile and they should do the right thing.

Now when I try to run make-pins.py, it cannot find stm32f4xx_af.csv.  There are also refs to other missing files, such as pins_af_const.h.

Also, there are refs to mpconfigboard.h.  There does not not seem to be one specifically for the OpenMV board.  Is that something I would also need?

iabdalkader, 2015-05-17:
you're not supposed to run anything, just type make in src/

waldo3000, 2015-05-17:
Sorry, I guess I managed to confuse what my current issue is.  I am asking about the missing files mentioned in my previous message.

iabdalkader, 2015-05-17:
Again, there are No missing files, those files are auto generated. you need to clone the repo:
```
git clone --recursive https://github.com/iabdalkader/openmv.git
```
And then move to src/ and type make
```
cd openmv/src/
make
```
That's it.

iabdalkader, 2015-07-09:
I assume this is answered, reopen if not.

closed 2015-07-09 by iabdalkader