openmv/openmv #49: IDE 1.1 will not run in windows 10

rkertesz, 2015-08-22:
Runs on windows 7 but not on win 10. 
Double clicking it doesn't pop up a single message box or anything but a spinning circle for 3 seconds. Nothing shows up in task manager.

iabdalkader, 2015-08-22:
Will test on Windows 10, in the meantime, could you try to run it from CMD ? see if it prints any messages.
Thanks!

rkertesz, 2015-08-26:
I tried running from command line. I simply executed the executable and it when to the next command prompt after thinking that it started the program. No error message.

iabdalkader, 2015-08-26:
Okay, will test it on Windows 10 and get back to you.

iabdalkader, 2015-08-28:
Tested on Windows 8 64-bit and Windows 10 64-bit:
[image]

[image]

rkertesz, 2015-08-28:
Hm. That is interesting.
Is there a prerequisite that needs to be installed? Maybe my computer (microsoft surface pro 3) was docked (I don’t recall)? That could have caused a video driver type issue?
I didn’t see it in the task tray either though. 
Are there any steps that i should follow other than double clicking the executable? Do i need to plug in the OpenMV beforehand?
I tried all sorts of different modes.
I have lots of services running too. Maybe my antivirus, maybe conflict with my installed python or other service?

iabdalkader, 2015-08-29:
No it just worked out of the box, it could be an antivirus.

kwagyeman, 2016-04-26:
Old.

closed 2016-04-26 by kwagyeman