openmv/openmv #50: Running openmv-ide.py on Mac OSX

blmorris, 2015-09-02:
I was able to get `usr/openmv-ide.py` to run on OSX - I don't have a board yet, just trying to figure out if it would be possible on my machine.

I just wanted to offer a few notes on meeting the dependencies to run the IDE on OSX - some of this may be relevant to other systems as well.

Using MacPorts (sudo port install …) I installed the following packages:
- python27
- py27-pip
- py27-numpy
- py27-pygtksourceview
- py-pygtk (will bring py27-pygtk along with it)
- gtk2, glade, and gtksourceview2 will probably get installed as dependencies if the above listed dependencies get installed; there may be others that I already had installed but these were the ones I needed to install specifically.

Then using pip (sudo pip-2.7 install …):
- pyserial
- Pillow (provides PIL - Python Imaging Library)

This took a little while to figure out, so I may have missed a few… I will add others as I think of them.
Note that many OSX users with well-established MacPorts and Python2.7 installations may not need to worry about any of this, or may only be missing a few things. My biggest problem was that the names of the outside packages that I needed were not always obvious.

iabdalkader, 2015-09-02:
Thank you for the info, I was planning on doing a package for OSX soon.

> EDIT- I assume that it is also necessary to install libusb (using MacPorts) and pyusb (using pip) to talk to the OpenMV board, but I don't have the board to confirm that yet.

Only if you want to use the built-in DFU loader.

blmorris, 2015-09-02:
@iabdalkader -
> Only if you want to use the built-in DFU loader.

Thanks, good to know. I do use the DFU loader for the pyboard (I believe this was originally your development?) but I wasn't sure exactly what the dependencies were. I can confirm that those parts work just fine on OSX.

iabdalkader, 2015-09-02:
Yes that's the same script (pydfu.py)

stleon, 2015-09-06:
We need a package for OS X! It will be nice

iabdalkader, 2015-09-06:
@stleon working on it

iabdalkader, 2015-09-07:
Got it working!

[image]

blmorris, 2015-09-08:
Very cool, looking forward to playing with it!

iabdalkader, 2015-09-08:
There's a Mac OSX package now:

https://github.com/openmv/openmv/releases/download/v1.1/openmv_darwin_x86_64_v1.1.zip

kwagyeman, 2016-04-26:
Old - but, we still need to make Mac IDE release.

closed 2016-04-26 by kwagyeman