openmv/openmv #54: Python standart libraries

Duiesel, 2015-11-18:
Hello Ibrahim!
I'm searching the libraries that You have been used in examples (in face_tracking.py)
```python
import sensor, time, image
```

As I know time-is the internal Python library, but others-no.  :( 
Can You help me: where can I find them?

Thank You very much!
Best Regards, 
Alan

iabdalkader, 2015-11-18:
Do you need the API documentation or modules code ?

Duiesel, 2015-11-21:
Hello, Ibrahim!
I need modules code: can't find it in the repo.

Documentation is good too, but isn't necessary.

iabdalkader, 2015-11-21:
https://github.com/openmv/openmv/tree/master/src/omv/py

Duiesel, 2015-11-21:
I'm sorry does it mean that importing in OpenMV IDE working with not only .py files?

AlexisTM, 2017-06-03:
http://docs.openmv.io/library/omv.sensor.html

What you were looking for was this, the documentation.

iabdalkader, 2017-06-03:
@Duiesel Sorry I must've missed your last comment. Yes you can import .py files but sensor, image and time are all builtin modules (embedded in the firmware). Note the time module is not exactly the same standard Python time module, it's best to always check the docs linked by @AlexisTM.

closed 2019-09-09 by iabdalkader