openmv/openmv #61: Add more sensor functions

kwagyeman, 2016-02-14:
Please add:

sensor.gain_control(off_or_on) // false/true to turn auto gain control (agc) off or on respectively.
sensor.white_balance(off_or_on) // false/true to turn auto white balance (awb) off or on respectively.
sensor.horizontal_mirror(off_or_on) // false/true to turn horizontal mirror off or on respectively.
sensor.vertical_flip(off_or_on) // false/true to turn vertical flip off or on respectively.

iabdalkader, 2016-02-15:
how about:
sensor.set_auto_function(sensor.AUTO_AWB, False/True) ?
There are so many of them to add one function for each auto function, (AWB, AGC, AEC, auto-denoise, auto-sharpness etc...)

iabdalkader, 2016-02-15:
And how do you want to set manual values ?

kwagyeman, 2016-02-15:
I don't know, I do know that most of those things have a single control bit that can be flipped. So, let's just start with that. Then maybe add another function which lets you control the values.

iabdalkader, 2016-03-25:
Implemented.

closed 2016-03-25 by iabdalkader