openmv/openmv #83: Make SD card faster

kwagyeman, 2016-02-26:
We're hitting the limit of single block mode with our current SD card setup which is about 250KBs. We should be able to reach 12.5MBs with the STM32. So, we're seriously under performing right now. The reason for this is because we have just a single 512 byte buffer. This limits us to reading and writing on sector at a time which SD cards are not really designed to do anymore and are slow at. If we want faster speeds we have to read/write multiple sectors at a time.

So, I propose we increase the buffer to 4KB and move it into the main RAM frame buffer area. We have plenty of space there and a new variable in the linker script can be created to reserved a 4KB chunk for it. I suggest we insert the thing in between the flash buffer and fb_alloc stack.

Doing the above will also allow use to free 512 bytes in the CCM ram area and let us use DMA mode for faster writes/reads.

I think we should see more than a 8X performance improvement by doing all the above.

naniBox, 2016-02-27:
Beware with SD card stalls though! Depending on the card, a stall can happen for up to 500ms, during this time, you will have to either buffer up the streamed data, lose it, or just block.

kwagyeman, 2016-02-27:
So, after looking at the code and checking a new things we're actually getting about 2MBs+. However you can only get that speed with grayscale images. I'll have to implement a large image buffer to make saving RGB565 images faster. (Grayscale images take 31ms to save 320x240 - so, 30 FPS).

We store RGB565 images byte reversed because of the order bytes are sent out of the OV cameras. Because of this we have to byte reverse all the pixels in RGB565 BMP images. This destroys the speed for them.

kwagyeman, 2016-02-29:
So, instead of an image buffer I will be byte reversing the images. I will get on that this week.

kwagyeman, 2016-03-22:
This has been fixed. It's now crazy fast. We buffer up huge writes in our extra RAM (24KB at a time) to get near 25Mb/s.

>> Beware with SD card stalls though! Depending on the card, a stall can happen for up to 500ms, during this time, you will have to either buffer up the streamed data, lose it, or just block.

IDE has a 2 second timeout for this to prevent from disconnecting.

closed 2016-03-22 by kwagyeman