openmv/openmv #88: Move LCD/VIDEO/BLE into modules that are builtin.

kwagyeman, 2016-02-29:
We can't have these be external scripts... way to unprofessional. We're selling an all in one package. I don't want any additional steps on using video.

That said, we don't need to support any more functionality that what the video script already does - but, it should be built-in.

iabdalkader, 2016-02-29:
Those scripts don't have to be built-in, if something doesn't have to be built-in it stays out to save flash, ram and to be easier to modify. This should more than enough to keep those drivers as scripts, but also keep in mind we're exposing low level peripherals to Python so folks could write drivers in Python and don't have to recompile firmware, those serve as examples for drivers, It's a feature not a bug ;)

kwagyeman, 2016-03-22:
Done.

closed 2016-03-22 by kwagyeman