openmv/openmv #116: JPEG decode error (cannot decode image data)

kidswong999, 2016-04-13:
hello,I made an openmv1 board, and use ov2640, but when I run helloworld.py in ide, the terminal always print "JPEG decode error (cannot decode image data)".

could you help me?

iabdalkader, 2016-04-13:
I'm not doing much work on OpenMV 1 right now since it was never released.

kidswong999, 2016-04-13:
so, OpenMV 1 will not update?  it's so tiny, and i want to add it to my mini quadrotor. 
And I want to join the project, is there any group?

iabdalkader, 2016-04-13:
I will get around to it, but OpenMV-2 has higher priority right now

kidswong999, 2016-04-13:
I think the error is, my ov2640 module is not the same as yours. so...could you send me your cam module infomation? Because of my skill level, I rewrite the driver is TOO difucult...
thanks!

kidswong999, 2016-04-17:
I save RGB565 image into SD card, but the image is mess up.
like this
[image]

could you help me about this?

kwagyeman, 2018-04-08:
Closing old stuff.

closed 2018-04-08 by kwagyeman