openmv/openmv #134: openmv2 Failed to init sensor, error:-2

kidswong999, 2016-05-01:
everything is OK, except the sensor.
OpenMV2 failed to init sensor, error:-2.
I search the source code, maybe can not probe sensor.

thanks.

closed 2016-05-02 by kidswong999

Iamttp, 2019-07-20:
my openmv4 produced the same problem "FATAL ERROR: Failed to init sensor, error:-2".
Can you tell me how to solve it in the end? Should I replace a new camera?

kwagyeman, 2019-07-20:
The OMV2 is a rather old system. Where I'd dyou buy it from.

Iamttp, 2019-07-20:
Hi, Can you tell me what should i do when my openmv4 produced the problem "FATAL ERROR: Failed to init sensor, error:-2"?

Iamttp, 2019-07-20:
Is it necessary to replace the camera?

kwagyeman, 2019-07-20:
Um, so, if you look at the source code sensor.c you can see what throws that error.

Um, generally, that means the hardy has issues. I'm not in a place to be able to answer this question in more detail right now.

iabdalkader, 2019-07-20:
It generally means the sensor is not responding to I2C scans. Did you remove/replace the sensor ?

Iamttp, 2019-07-20:
no, I have been using it all the time. The camera was still good before

Iamttp, 2019-07-20:
The last time I used it was to use the dupont line give openmv five volts. and the LED kept shining and i found a error log file.

iabdalkader, 2019-07-20:
dupont line ?

Iamttp, 2019-07-21:
I am so sorry to my English. It just powers openmv.

iabdalkader, 2019-07-21:
Anyway, this is not a software issue, it sounds like the camera has been damaged and you'll need a replacement.

Iamttp, 2019-07-22:
OK, thank you