openmv/openmv #136: Hardware: Use through Micro-USB Connector

kwagyeman, 2016-05-18:
For new OpenMV Cams the micro-USB connector needs to use a through-hole variant to make it more robust. I realize that a through-hole connector will require the servo headers on the back of the board to be dropped. This is fine. Those connectors prevent MacroFab from using a selective solder machine easily which increases the cost of building the boards: https://www.youtube.com/watch?v=p-VImd2yW5s

kwagyeman, 2016-05-18:
See the connector the pyboard uses: https://github.com/micropython/pyboard/tree/master/eagle

anedel, 2016-06-06:
@kwagyeman: You wrote "a through-hole connector will require the servo headers on the back of the board to be dropped". Why is it required?

kwagyeman, 2016-06-06:
They run under the USB header.

closed 2017-05-27 by kwagyeman