openmv/openmv #139: Is release 1.5 compatible with older cams?

rkertesz, 2016-06-04:
I have two of the old alpha/pseudo beta cams that were on tindie. What is the latest release that is still compatible with those? Any special instructions for us?

iabdalkader, 2016-06-06:
Hi, the build for OMV1 has broken along the way, never got around to fixing it, the firmware now requires many fixes to support OMV1, if you have an OMV2 you should use it instead, otherwise if there's more demand for supporting OMV1 I'll fix it.

rkertesz, 2016-06-29:
I may be the only one who could really use this but I have two of the OpenMV R5 cams (I believe this is openmv1). Which build is still compatible with my camera. If I am the only one, then I guess I can make due with an older release. Thanks!

rkertesz, 2016-06-29:
I am only able to use version 1.1 of the IDE. I tried versions 1.5 and 1.4. They said I needed to update the firmware. When I used the bootloader tool, it said unplug the camera and plug it back in. When I reconnected the camera, the message didn't go away. Version 1.3 and 1.2 didn't appear to have a bootloader integrated into IDE. If I was to upgrade the firmware, would I have to short the pins to set it into DFU mode?

rkertesz, 2016-06-29:
Edit:doesn't work in dfu mode. I am upgrading using dfuse util

rkertesz, 2016-06-29:
This is probably of no surprise to anyone visiting this issue but, even though I upgraded to the firmware using the DFU file in the openmv1 directory of the v1.5 release, the IDE stated that the firmware was old and must be updated. I then checked in version 1.2 to see if it worked there and it did not. Is it therefore safe to conclude that there is no compatibility at this point past version 1.1 for both of my openmv1 cameras? Thanks! :)

iabdalkader, 2016-06-29:
Hi, the firmware doesn't support OpenMV1 anymore, if there's more demand to fix it I will, in the mean time you could pre-order OMV2 it's a much better cam in every way.

rkertesz, 2016-06-29:
Thank you. I am in the kickstarter. Hopefully we will receive our cams soon :)
I would suggest removing the OpenMV1 folder from future releases just in case there is someone else out there with the same issue.

mindThomas, 2016-07-18:
I have 4x OpenMV1 cams already and would be very happy if the build was working too as I was actually planning on doing a writeup about these.
As my project requires all four cameras it would be quite an expense for me to go out and buy another four of the new cameras, why support for the firmware would be good.

kwagyeman, 2016-08-16:
It's not really possible t support the old hardware. We use a lot more flash and RAM than the old board supports. So, we'd have to cut features down and do a lot of IFDEF work to get that operational.

Since we don't have a lot of man power we won't be working on supporting the old boards.

x893, 2016-08-17:
I use STM32F407VG with OpenMV1 and last software version with small fixes.
see detail here https://github.com/x893/openmv
(because PA9 damage on my module - USB detection disabled in codes)

kwagyeman, 2018-04-08:
Closing old stuff.

closed 2018-04-08 by kwagyeman