openmv/openmv #147: Allow configuration of FIR refresh rate & ADC resolution

wminarik, 2016-08-04:
(for the Thermopile Shield)

Currently within py_fir.c the MLX90621 initialization is set to:
64 Hz refresh rate
18-bit ADC resolution

I suggest allowing these parameters to be set on the call to fir_init(), either by adding arguments to this function or by adding a new function (fir_config()?) that takes a valid refresh rate (0.5 to 512 Hz) and a valid ADC resolution (15-18 bits).

Reasoning: the noise level of MLX90621 corner pixels decreases from ~ 1°C at 64 Hz to less than 0.1°C at 1 Hz. When the frame rate is slower than 64 Hz, the thermal images are noisier than need be. Stationary objects would benefit from a lower noise floor.

The current 18-bit ADC resolution gives the smallest temperature increment within range, but limits the maximum object temperature to ~450°C. Objects within the field hotter than this may return invalid data due to measurement overflow (according to the datasheet). There may be applications in which it's desired to have hotter objects within the field of view (up to a maximum of ~950°C with a 15-bit ADC resolution setting).

kwagyeman, 2016-08-16:
Yes, we can easily do this. However, it you want it done sooner than later can you submit a code patch for this feature? I wrote the MLX code such that all the calcs are done during the init time. It shouldn't really be that much work to add this feature in.

Otherwise... this is rather low priority for me and it won't get done for a long time.

iabdalkader, 2018-04-26:
Closing this because we're not supporting FIR MLX anymore.

closed 2018-04-26 by iabdalkader