openmv/openmv #152: Camera crashes when IDE overwrites the currently running script

petri, 2016-09-13:
(originally https://github.com/openmv/openmv-ide/issues/5)

Steps to reproduce:

1. connect camera
2. start a script on it, from camera SD card
3. while the script is still running, make some changes and re-save (via IDE), ie. overwrite the script currently running on the camera

There will usually be different errors, apparently depending on the actual changes to the script. The only way to recover is to physically disconnect-reconnect the USB.

The crash does not always happen - it seems the more there are changes to the script, the higher the likelihood of crash.

iabdalkader, 2017-07-18:
@kwagyeman Is this related to the issue you were talking about ?

kwagyeman, 2018-04-08:
Old issue.

closed 2018-04-08 by kwagyeman