openmv/openmv #153: Kickstarter cams arent working with latest IDE [Ubuntu 16.04]

twitchyliquid64, 2016-09-22:
Hi,

I downloaded the latest and followed the instructions to set them up. When I plug in the cam, it flashes green, then magenta, then the LED goes out.

The camera shows up in /dev/openmvcam and mounts as a storage device.

When I click the connect button in the IDE, I am told I _must_ update my camera firmware. I hit OK and it prompts me whether I want to erase internal filesystem.

Regardless of which I choose, the LED on the board flashes the same sequence as above then says 'Unable to connect to bootloader!' I have tried many times with exactly the same result.

What should I do?

twitchyliquid64, 2016-09-22:
```

[xxx@xxx-MS-7758]:/usr/share> lsusb
Bus 002 Device 019: ID 1209:abd1 InterBiometrics 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 2516:003b  
Bus 001 Device 003: ID 046d:c21f Logitech, Inc. F710 Wireless Gamepad [XInput Mode]
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1c4f:0003 SiGma Micro HID controller
Bus 003 Device 002: ID 046d:0829 Logitech, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

twitchyliquid64, 2016-09-22:
```
> uname -a
Linux xxx-MS-7758 4.4.0-36-generic #55-Ubuntu SMP Thu Aug 11 18:01:55 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

twitchyliquid64, 2016-10-03:
Updated IDE to latest version. Same result.

This is the system log - it shows some errors: are these expected?

```
Oct  3 21:10:20 xxx-MS-7758 kernel: [37810.720065] cdc_acm 1-1.2:1.0: failed to set dtr/rts
Oct  3 21:10:20 xxx-MS-7758 kernel: [37810.734836] cdc_acm 1-1.2:1.0: failed to set dtr/rts
Oct  3 21:10:20 xxx-MS-7758 kernel: [37810.957300] usb 1-1.2: USB disconnect, device number 20
Oct  3 21:10:20 xxx-MS-7758 kernel: [37810.959556] cdc_acm 1-1.2:1.0: failed to set dtr/rts
Oct  3 21:10:20 xxx-MS-7758 ModemManager[1009]: <info>  (tty/ttyACM0): released by modem /sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2
Oct  3 21:10:20 xxx-MS-7758 udisksd[2083]: Cleaning up mount point /media/xxx/F26C-5520 (device 8:49 no longer exist)
Oct  3 21:10:20 xxx-MS-7758 kernel: [37811.189653] usb 1-1.2: new full-speed USB device number 21 using ehci-pci
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.284124] usb 1-1.2: New USB device found, idVendor=1209, idProduct=abd1
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.284126] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.284128] usb 1-1.2: Product: OpenMV Virtual Comm Port in FS Mode
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.284129] usb 1-1.2: Manufacturer: OpenMV
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.284130] usb 1-1.2: SerialNumber: 000000000011
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.284419] cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device
Oct  3 21:10:21 xxx-MS-7758 mtp-probe: checking bus 1, device 21: "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2"
Oct  3 21:10:21 xxx-MS-7758 mtp-probe: bus: 1, device: 21 was not an MTP device
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.469414] usb 1-1.2: USB disconnect, device number 21
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (ttyACM1) could not open serial device (2)
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (Plugin Manager) (Nokia) [ttyACM1] error when checking support: '(tty/ttyACM1) failed to open port: Could not open serial device ttyACM1: No such file or directory'
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (ttyACM1) could not open serial device (2)
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (Plugin Manager) (Via CBP7) [ttyACM1] error when checking support: '(tty/ttyACM1) failed to open port: Could not open serial device ttyACM1: No such file or directory'
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (ttyACM1) could not open serial device (2)
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (Plugin Manager) (Cinterion) [ttyACM1] error when checking support: '(tty/ttyACM1) failed to open port: Could not open serial device ttyACM1: No such file or directory'
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (ttyACM1) could not open serial device (2)
Oct  3 21:10:21 xxx-MS-7758 ModemManager[1009]: <warn>  (Plugin Manager) (Generic) [ttyACM1] error when checking support: '(tty/ttyACM1) failed to open port: Could not open serial device ttyACM1: No such file or directory'
Oct  3 21:10:21 xxx-MS-7758 kernel: [37811.471659] cdc_acm 1-1.2:1.0: failed to set dtr/rts
Oct  3 21:10:21 xxx-MS-7758 kernel: [37812.177646] usb 1-1.2: new full-speed USB device number 22 using ehci-pci
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.272393] usb 1-1.2: New USB device found, idVendor=1209, idProduct=abd1
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.272398] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.272401] usb 1-1.2: Product: OpenMV Virtual Comm Port in FS Mode
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.272403] usb 1-1.2: Manufacturer: OpenMV
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.272404] usb 1-1.2: SerialNumber: 000000000011
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.272802] cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.273210] usb-storage 1-1.2:1.2: USB Mass Storage device detected
Oct  3 21:10:22 xxx-MS-7758 mtp-probe: checking bus 1, device 22: "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2"
Oct  3 21:10:22 xxx-MS-7758 mtp-probe: bus: 1, device: 22 was not an MTP device
Oct  3 21:10:22 xxx-MS-7758 kernel: [37812.274587] scsi host14: usb-storage 1-1.2:1.2
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.274507] scsi 14:0:0:0: Direct-Access     uPy      microSD SD card  1.00 PQ: 0 ANSI: 2
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.275160] sd 14:0:0:0: Attached scsi generic sg4 type 0
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.275696] sd 14:0:0:0: [sdd] 6799360 512-byte logical blocks: (3.48 GB/3.24 GiB)
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.276320] sd 14:0:0:0: [sdd] Write Protect is off
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.276325] sd 14:0:0:0: [sdd] Mode Sense: 00 00 00 00
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.276984] sd 14:0:0:0: [sdd] Asking for cache data failed
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.276989] sd 14:0:0:0: [sdd] Assuming drive cache: write through
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.285595]  sdd: sdd1
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.285601] sdd: p1 size 15179776 extends beyond EOD, enabling native capacity
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.293353]  sdd: sdd1
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.293358] sdd: p1 size 15179776 extends beyond EOD, truncated
Oct  3 21:10:23 xxx-MS-7758 kernel: [37813.296944] sd 14:0:0:0: [sdd] Attached SCSI removable disk
[... cut ...]
Oct  3 21:10:26 xxx-MS-7758 systemd[1]: Started Hostname Service.
Oct  3 21:10:37 xxx-MS-7758 ModemManager[1009]: <info>  Creating modem with plugin 'Generic' and '1' ports
Oct  3 21:10:37 xxx-MS-7758 ModemManager[1009]: <warn>  Could not grab port (tty/ttyACM1): 'Cannot add port 'tty/ttyACM1', unhandled serial type'
Oct  3 21:10:37 xxx-MS-7758 ModemManager[1009]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2': Failed to find primary AT port
```

iabdalkader, 2016-10-03:
Could you try updating the firmware with DFU ?

iabdalkader, 2017-07-18:
So we have a new IDE now and new FW. If you're still having problems with the IDE + FW reopen the issue.

closed 2017-07-18 by iabdalkader