openmv/openmv #171: stm32f767zi Firmware download

846354866, 2017-01-12:
I downloaded the 'bootloader.bin' and 'firmware.bin' to the NUCLEO-F767ZI board, but the board did not run. I am confused, I hope to get your help.

kwagyeman, 2017-01-12:
Hi, the code only runs on the OpenMV Cam. Buy one here: https://openmv.io/collections/products/products/openmv-cam

closed 2017-01-12 by kwagyeman

846354866, 2017-01-13:
thank you!
I would like to ask:
Do you intend to open source to the firmware in the future? Or can I use a third-party development board?

kwagyeman, 2017-01-13:
This repo is the firmware source... and it's open...

We  won't be supporting any third party dev board. The OpenMV code based is developed and tested for the OpenMV Cam. If you want to try something else with it that's on you.

846354866, 2017-01-16:
Hi, I have the following error when compiling the firmware:
AS src/st/startup_stm32f765xx.s
src/st/startup_stm32f765xx.s: Assembler messages:
src/st/startup_stm32f765xx.s:47: Error: unknown cpu `cortex-m7'

The version of gcc-arm-none-eabi is v5.4.1.
I ask, where I made a mistake?

kwagyeman, 2017-01-16:
Hi, I'm not going to provide you help support for this. I don't know what's causing the error you are getting. If you're building OpenMV for another platform you are on your own. Please google for unknown cpu `cortex-m7'.