openmv/openmv #205: Compilation error

terrygarbie, 2017-04-03:
Hi,
``
I referenced the webpage https://github.com/openmv/openmv/wiki/Firmware-Development, and make the firmware.
I got the compilation errors as below:
CC src/dsp/MatrixFunctions/arm_mat_add_q15.c
src/dsp/MatrixFunctions/arm_mat_add_q15.c: In function 'arm_mat_add_q15':
src/dsp/MatrixFunctions/arm_mat_add_q15.c:109:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
       *__SIMD32(pOut)++ = __QADD16(*__SIMD32(pInA)++, *__SIMD32(pInB)++);
       ^
src/dsp/MatrixFunctions/arm_mat_add_q15.c:109:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
src/dsp/MatrixFunctions/arm_mat_add_q15.c:109:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
src/dsp/MatrixFunctions/arm_mat_add_q15.c:110:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
       *__SIMD32(pOut)++ = __QADD16(*__SIMD32(pInA)++, *__SIMD32(pInB)++);
       ^
src/dsp/MatrixFunctions/arm_mat_add_q15.c:110:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
src/dsp/MatrixFunctions/arm_mat_add_q15.c:110:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
cc1: all warnings being treated as errors
make[1]: *** [/home/terrychan/workspace_stm23/openmv/src/build/cmsis/src/dsp/MatrixFunctions/arm_mat_add_q15.o] Error 1
make[1]: Leaving directory '/home/terrychan/workspace_stm23/openmv/src/cmsis'
make: *** [FIRMWARE_OBJS] Error 2

Can anyone help?
Thanks!

Terry

iabdalkader, 2017-04-03:
Hi,
I think you have an old toolchain, what's the gcc version ?

'arm-none-eabi-gcc -v'

The last tested toolchain is 2016-q3  you can find it here:
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

terrygarbie, 2017-04-04:
I am using gcc in version as below.
gcc version 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512] (GNU Tools for ARM Embedded Processors 6-2017-q1-update)

iabdalkader, 2017-04-04:
2016-q3 is gcc 5.4 maybe newer gcc issue warnings for that code, could you try adding `-Wno-error=strict-aliasing` to the main Makefile CFLAGS?
`CFLAGS += -std=gnu99 -Wall -Werror -Warray-bounds -mthumb -nostartfiles -mabi=aapcs-linux -fdata-sections -ffunction-sections -Wno-error=strict-aliasing
`

terrygarbie, 2017-04-05:
Hi iabdalkader,
It works. Thanks.
But is it a workaround?

iabdalkader, 2017-04-05:
Yes for now, it stops that specific warning from being made into an error, note that DSP code is not actually used and it's STM code, next time I update the toolchain I'll fix it anyway.

closed 2017-04-05 by terrygarbie