openmv/openmv #207: Why not apply stmhal in src/micropython/stmhal folder?

rmokerone, 2017-04-07:
In this project, I found  two ``stmhal`` folder. On in the ``src/stmhal`` path and the other under the ``src/micropython/stmhal/hal`` path. They look only different version number. So, why not use stmhal in the ``micropython`` folder?  If so, maybe it would be better to synchronize with ``micropython``.

iabdalkader, 2017-04-08:
Things change a lot in micropython, a change in STM could break something which I have to track for hours (or days) to figure out. Plus not every change is actually needed for OpenMV. Keeping our own versions of libraries is much better for abstraction and tracking changes. (note there's also CMSIS, FatFS etc.. outside micropython).

rmokerone, 2017-04-08:
Thank you very much for your answer. Although it will be a lot of trouble. But ``micropython`` supports a lot of devices. If OPENMV can better synchronize with ``micropython``, maybe OPENMV can run better on other ``micropython`` supported devices? For example, ``micropython`` is now doing ESP32 porting, if synchronized, perhaps ``openmv`` will be able to run on the ESP32 chip?
PS:Do you have plans to run openmv on ESP32??

iabdalkader, 2017-04-08:
There are no plans to run on ESP right now, but using stmhal inside MP wouldn't help because ESP will have different drivers, HALs and will require new board support code.

closed 2017-04-25 by iabdalkader