openmv/openmv #208: about WiFi Shield

fly-123, 2017-04-08:
hello,I have a question for WiFi Shield,used OpenMv IDE-Examples-WiFi Shield-scan.py can't output anything on Serial Terminal. i have connect Cam M7 and WiFi Shield.
why this problem? thks!

iabdalkader, 2017-04-08:
Does it print the firmware version ? If not this means the shield is not connected all the way, sometimes pins don't make full contact. If it's connect and prints the firmware version then there's no hardware problem and it should print nearby public access points names.

fly-123, 2017-04-08:
thank you ! so, soft code is correct. I will find  hardware problem.now, led D1 Light up is very weak on power supply. i think,this is a Abnormal phenomenon?

fly-123, 2017-04-08:
ye! The WiFi Shield is also work. I have try do this of update the Cam M7 firmware version use dfu.

closed 2017-04-09 by fly-123

iabdalkader, 2017-04-09:
Note LED D1 is floating when the module is reset, so it's normal.

fly-123, 2017-04-09:
yes! I have also used WiFi Shield for MJPEG Streaming AP model complete.
this is very good! thanks for your Shared!