openmv/openmv #212: i2c in OpenMV3

kidswong999, 2017-04-20:
I connected PCA9685 servo module to openmv, and I found a bug.
i2c in OpenMV2:
```
>>>from pyb import I2C
>>>i2c = I2C(2, I2C.MASTER)
>>>i2c.scan()
[64, 112]
>>>i2c.mem_read(1,64,1)
b'\x04'
>>> i2c.mem_read(1,64,0)
b'\x11'
```
i2c in OpenMV3:
```
>>>from pyb import I2C
>>>i2c = I2C(2, I2C.MASTER)
>>>i2c = I2C(2, I2C.MASTER)
>>>i2c.scan()
[64, 112]
>>> i2c.mem_read(1,64,1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 116
>>> i2c.mem_read(1,64,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 116
```

Is there an issue in the firmware?

zz2633105, 2017-04-25:
I have also encountered the same problem

iabdalkader, 2017-04-25:
Hi, I don't have that module, but I'll test I2C and get back to you.

zz2633105, 2017-04-27:
Hey, did the test results come out?

kwagyeman, 2017-04-27:
Hi, please post issues to the forum. As for the above problem. Try using the non-memory commands. Like i2c.send() and i2c.recv(). I have some good example scripts on how to use I2C packaged with the IDE. Checkout the Pixy emulation scripts in particular. Also, you need to have pull-up wires on the I2C lines.

iabdalkader, 2017-04-27:
@zz2633105 @kidswong999 Please let me know if the issue was missing resistors if not I'll debug the code.

zz2633105, 2017-04-28:
@iabdalkader I do not know if the lack of resistance, but the same operation, OpenMV2 can run, OpenMV can not. This is why?
[image]

zz2633105, 2017-04-28:
@iabdalkader PCA9685模块有上拉电阻

iabdalkader, 2017-04-28:
I think you need pull-ups, I don't see any resistors on that board. Do you have a link/pic of the PCA board ?

zz2633105, 2017-04-28:
@iabdalkader The PCA9685 module has a pull-up resistor

iabdalkader, 2017-04-28:
You should try this [library](https://github.com/adafruit/micropython-adafruit-pca9685/blob/master/pca9685.py) and import i2c from machine not pyb, as suggested in the forums.

zz2633105, 2017-04-28:
@iabdalkader alright, thank you very much

kidswong999, 2017-04-28:
Hello, I am using machine instead of pyb, and it works. 
Thanks. : )

kidswong999, 2017-04-28:
I upload code and tutorial at http://openmv.cc/doku.php?id=tutorial:7
I will be uploaded to the forum recently.
[openmv-pca9685.zip](https://github.com/openmv/openmv/files/963384/openmv-pca9685.zip)

closed 2017-04-28 by kidswong999

iabdalkader, 2017-05-14:
@kidswong999 Hey would you be interested in re-writing those tutorials in English ?  We're looking for someone to help with tutorials. If you're interested contact me at iabdalkader at openmv.io