openmv/openmv #214: Is any intuitive way to get the image content？

rmokerone, 2017-04-25:
First I get the data from openmv and send it via spi.But I did not find any way to get data directly.So I did that:
```python
    img = sensor.snapshot() # Take a picture and return the image.
    jpg = img.compressed()
    jpg.save('data.jpg')
    f = open('data.jpg', 'rb')
    data = f.read()
    f.close()
    data = b'start'+data+b'end'
    spi.send(data)
```
Is there any other more elegant way to do this thing?
Thanks~

iabdalkader, 2017-04-25:
There's a [get_pixel](http://docs.openmv.io/library/omv.image.html?highlight=get_pixel#image.image.get_pixel) function for raw images. Is this what you want ?

rmokerone, 2017-04-26:
I have seen this function.But this function. But this function can only get one pixel at a time. If I want to get all the pixels to the entire image. You have to call this function N times. Obviously, this efficiency will be relatively low. So, is there a function to get all the pixels of the entire image at once?

iabdalkader, 2017-04-26:
No this is the only function to access pixels, if I know more about what you're trying to do and why maybe I can help.

rmokerone, 2017-04-27:
I want to send the image to the computer via ``ESP8266``. 
>The first step is to get the image through ``OpenMV``. 
>The second step, the image sent to ``ESP8266`` through ``SPI``. 
>The third step, ``ESP8266`` will send the image to the computer.
 I did this, but it was slower, so I wanted to optimize the process. 

Note: I know ``WINCE`` suite, but I want to try to do with ``ESP8266``. If there are any suggestions, would be very grateful.

iabdalkader, 2017-04-27:
You don't need to access the raw pixels for that, you just need to compress the image and then pass the image object to the socket send function:
```
  img = sensor.snapshot()
  img = img.compressed(quality=35)
  sockobj.send(img, ...)
```
However, there's no Python module for ESP which implements the sockets layer functions (for example see WINC's Python [module](https://github.com/openmv/openmv/blob/master/src/omv/py/py_winc.c)).

But you can still use ESP8266 as a serial (or SPI) device, see for example the BLE [script](https://github.com/openmv/openmv/blob/master/usr/examples/13-BLE-Shield/ble.py)

In this case you should use `uart.write(img)`

closed 2017-07-06 by iabdalkader