openmv/openmv #225: Image library issued

Mecatronique, 2017-05-02:
Hello, 

I am trying to run the **template_matching** programme from the OpenMV IDE examples. 
When I launch it, I receive the following warning: **"ImportError: cannot import name SEARCH_EX"**. 
I tried to look up to know more about this SEARCH_EX algorithm and I am not quite sure about how to import it. 
It is supposed to be in the image module but my computer cannot find it. I feel like I do not have access to the image librabry. I also received **"AttributeError: 'module' object has no attribute 'SEARCH_EX' "**
I have an OpenMV3. 

Thank you for your help

iabdalkader, 2017-05-06:
Hi,

You just need to update the firmware.

Mecatronique, 2017-05-09:
Hi,

Thank you very much ! Now that works. Still, the entire sketch does not work. What we've done: 
  - taking a picture with OMV cam in grayscale, and saving it on the computer
  - pasting it in the camera directory (either in jpg, jpeg or bmp)
  - triying to use it as a template.
When we try to run the sketch, we get an error saying "**This function is only supported on GRAYSCALE images**". It points at all the **image** functions we try to use (like find_template).
Is it something we did wrong with the template?

Thanks for your reply

iabdalkader, 2017-05-09:
Yes you should set the sensor output to GRAYSCALE and use a PGM template.

Mecatronique, 2017-05-16:
Thank you very much for your fast answer ! 
But now I've got another problem : line 35-36 in code below, I get "FB : Alloc collision"
Is it a memory problem ?
I am also working on a serial port, but in the **I2C** function argument, what sould I put as an adress? Just addr=0x12 ?
[template_matching_modified.zip](https://github.com/openmv/openmv/files/1004312/template_matching_modified.zip)

iabdalkader, 2017-05-23:
Sorry for the delay, lines 35-36 are comments but yes FB alloc collision means you're running out of memory, probably using a big template.

Re I2C not sure what you're question is, but you should use the forums for project questions & help.

closed 2017-07-06 by iabdalkader