openmv/openmv #227: Light bleeding

Sejl, 2017-05-07:
Light bleeds through via holes on the back of the image sensor.

iabdalkader, 2017-05-09:
Hi,
The vias are all tented (painted over with the solder stop mask) which helps with the light bleeding but obviously doesn't completely stop it. I don't know of any other way to fix this (note cameras typically come in enclosures). If you have any **practical** suggestions (i.e. something we can do during manufacturing) please let me know and I'll forward to our factory.

Sejl, 2017-05-09:
Hi,

Sorry, but other than standard tips ([for example here](https://www.pcbuniverse.com/pcbu-tech-tips.php?a=5)) I have no other suggestions for you. I have solved problem by painting over vias.

scm6079, 2017-05-20:
This may be worth adding to the FAQ -- I was fighting terrible light bleed to the point vision tasks were impossible until simply adding a small piece of electrical tape over the back of the circuit board directly behind the image sensor area.

kwagyeman, 2017-05-23:
We're going to add black stickers to the back of all OpenMV cameras for our next production run. For the current ones a piece of electrical tape will do.

iabdalkader, 2017-06-07:
Hi, so we already use tented vias (one of the suggestions in the link above) and we're going to add the black stickers like @kwagyeman mentioned.

Additionally, we're in the process of designing a case. We were going to use transparent/clear plastic, but maybe now we should use an opaque color ?

lorth, 2017-07-19:
Use plugged vias: https://www.pcbuniverse.com/pcbu-tech-tips.php?a=5

kwagyeman, 2017-08-12:
Fixed with new black sticker on back.

closed 2017-08-12 by kwagyeman