openmv/openmv #229: openmv3  ADC function

846354866, 2017-05-11:
Hi, there is a problem that has been confusing for a long time, I want to achieve more than two ADC for data sampling, but I try to modify some of the source code, but still failed. Now I have no idea, I hope to get your help.

iabdalkader, 2017-05-12:
You mean use two ADC channels ? Unfortunately there's only one channel broken out on the header.

846354866, 2017-05-12:
@iabdalkader 
Thank you very much for your answer! Yes, I would like to use two ADC channels, or I would like to use another channel (not just channel 5).

iabdalkader, 2017-05-14:
There's only **1** channel broken out, sorry.

846354866, 2017-05-15:
Thank you! I have found other effective ways. There is an adc_init_all () function, I added some content, and now can achieve multiple channels data sampling.

iabdalkader, 2017-05-15:
Okay glad you found a way to do what you want :) going to close this issue now, if there's something else please open another issue. If you have questions please use the forums.

closed 2017-05-15 by iabdalkader