openmv/openmv #230: face detector details

PVNathan, 2017-05-15:
Hi,
I couldn't find the details regarding how the haar cascade is performed for the face detector. Can i know which file the actual math resides in? In face detection.py I see import image and then it says image.haarcascade but I couldn't ind where the image module is. I basically want to know how it detects the faces in detail. Appreciate your response. 
Thanks

iabdalkader, 2017-05-15:
Here:
https://github.com/openmv/openmv/blob/master/src/omv/img/haar.c

PVNathan, 2017-05-15:
Thanks. 
Also when I run the face_eye_detector.py it says no module named sensor. I am new to python and is there a readme where it says how to run the example py files?
Thanks

iabdalkader, 2017-05-15:
Those modules are available on OpenMV only, the code will not work on your PC.

PVNathan, 2017-05-15:
okay, thanks.

closed 2017-05-16 by iabdalkader