openmv/openmv #236: I cannot open openmv as a udisk while debuging in IDE is OK

jsbyysheng, 2017-05-30:
Hello,

It's very strange to use openmv3 in windows10 pro ver15063. In Device Manager, all the drivers work well. At the same time, I can also use IDE to debug my program. But I cannot open openmv3 as a udisk unless I wait for more than 10 minutes. Then I try to connect it to my Ubuntu16.04.1 virtual machine and all is ok...

When I connect it to my computer, I can hear the sound that openmv3 "connected -> disconnected ->connected".Also, I copy a program with "usb_vcp.isconnected()" in, I can see the led shows openmv3 "connected(Green) -> communicating(White) ->disconnected(Red)".(Green is connected and Red is discon).But IDE debugger always works well.

Finally, when I connect openmv3 to my computer, then plug in other usb devices, they have no response until I remove openmv3. Also, if I restart my computer with openmv3 plugged in, my computer will stop in startup state unless I remove openmv3.

Thank you for your help!

kwagyeman, 2017-05-30:
Yeah, I just encountered this same issue. I have no idea what's going on.

iabdalkader, 2017-05-30:
Hi seems a recent update broke something, I'm looking into it.

Please open an new issue (or edit this one) and the follow the issue **template** (how to reproduce, expected vs. actual behavior, OS, FW, IDE version all that) so I can help you more.

kwagyeman, 2017-05-30:
I also have a Ubuntu VM and it seems to work fine on that.

>> Finally, when I connect openmv3 to my computer, then plug in other usb devices, they have no response until I remove openmv3. Also, if I restart my computer with openmv3 plugged in, my computer will stop in startup state unless I remove openmv3.

Didn't notice this part, but, I bet I'm having the same issue.

We've cleared all other software issues for the next firmware release so I guess this will be our sole focus for the release. This issue kinda makes the system unusable.

kwagyeman, 2017-05-30:
Yeah, I have the same 15063.296 OS build. I think windows changed some stuff on us in the driver level when they forced that system update a few days back. Me thinks all pyboards are broken too.

jsbyysheng, 2017-05-30:
So sad to meet this problem. I'll edit this issue a little while. Thank u for paying attention to this issue.

kwagyeman, 2017-05-31:
Seems to be the Windows 10 Creator update. Going to try playing around with driver files and see if this helps.

jsbyysheng, 2017-05-31:
> Seems to be the Windows 10 Creator update. Going to try playing around with driver files and see if this helps.

playing around with driver files...I'm afraid it will result in disastrous consequence. 2333333

kwagyeman, 2017-05-31:
Not Windows drivers. But, the ones we create for hardware. I.e. our INF files.

kwagyeman, 2017-06-03:
MP fixed this issue. We'll release a new firmware with a patch for this soon.

commit 8cfa8f8b (2017-06-04): Apply SCSI layer fix from MP upstream.

* Fix #236
closed 2017-06-04 by iabdalkader