openmv/openmv #240: UART works not well.

jsbyysheng, 2017-06-06:
### Steps to reproduce
1.Connect UART3 to my PC by the UART to USB convertor.
2.Use uart.write("Hello World!\r")

### Expected behaviour
I want to get "Hello World!"

### Actual behaviour
I only get "Hello"

### Describe your setup

**Operating system**:
Windows 10
**OpenMV IDE version:**
1.6.0
**OpenMV firmware version:**
2.4.1

kwagyeman, 2017-06-06:
Hi, can you post the script you are using. The serial port was working fine the last time I used it. Something could have broke.

iabdalkader, 2017-06-06:
Please try "\n" instead of "\r" or "\r\n"

kwagyeman, 2017-06-07:
This is legit. The UART doesn't work for my script anymore. Ibrahim can you debug? See board_control/uart_control.py.

When you enabled UART 1 you may have broken this.

jsbyysheng, 2017-06-07:
> Hi, can you post the script you are using. The serial port was working fine the last time I used it. Something could have broke.

I use the example program as this:
```
import time
from pyb import UART
uart = UART(3, 19200)

while(True):
    uart.write("Hello World!\r")
    time.sleep(1000)
```

> Please try "\n" instead of "\r" or "\r\n"

Have no bussiness with this. I have tried uart.write("HelloWorld!") and I still receive "Hello".

kwagyeman, 2017-06-07:
Yeah, something is definitely broken. When I tried the same script I got the same result. Ibrahim enabled serial port 1 on the OpenMV Cam M7 which seems to have broke this. Interestingly, serial port 1 doesn't work either.

jsbyysheng, 2017-06-07:
I'm sorry to say the micropython is so unstable......

kwagyeman, 2017-06-07:
Ibrahim will fix this in a bit.

iabdalkader, 2017-06-07:
This issue is Not related to UART1. I disabled UART1 and tested, it's still there. It's Not related to any recent change either (the issue exists in **r2.3**). It seems that UART just timeouts, very short time_char for slow baudrates (<=19200 default timeout_char is 1ms). Setting **timeout_char=10** makes it work fine. I'll increase the default UART timeout.

Just use this for now:
`uart = UART(3, 19200, timeout_char=10)`

kwagyeman, 2017-06-07:
Ah yes, I remember this being an issue. MP is weird with this. I have no clue why they designed the UART with a timeout on the whole string.

iabdalkader, 2017-06-07:
The STHAL libraries accept timeout for the string and for each character (between characters). The character timeout is the issue. BTW I think this is standard, termios works the same. Anyway, closing this issue for now.

closed 2017-06-07 by iabdalkader

kwagyeman, 2017-06-07:
Maybe update the script?

iabdalkader, 2017-06-07:
I increased the default timeout_char.
https://github.com/openmv/openmv/commit/a5b25bd5b72311b43a9e3c59e6448c549b1b57c1

jsbyysheng, 2017-06-08:
OK