openmv/openmv #252: Cascade converter broken

normen, 2017-07-06:
### Steps to reproduce
1. Download cat haar cascade from here: https://github.com/opencv/opencv/tree/master/data/haarcascades
2. Download converter from here: https://github.com/openmv/openmv/blob/master/usr/openmv-cascade.py
3. Run converter on file

### Expected behaviour
A binary haar cascade file for the OpenMV cam is created

### Actual behaviour
The converter fails with this trace:
```
./openmv-cascade.py haarcascade_frontalcatface.xml
Traceback (most recent call last):
  File "./openmv-cascade.py", line 257, in <module>
    main()
  File "./openmv-cascade.py", line 254, in main
    cascade_binary(args.file, args.stages, args.name)
  File "./openmv-cascade.py", line 75, in cascade_binary
    size = (map(int, xmldoc.getElementsByTagName('size')[0].childNodes[0].nodeValue.split()))
IndexError: list index out of range
```

### Describe your setup

**Operating system**: MacOS

**OpenMV IDE version:** 1.6.0

**OpenMV firmware version:** 2.4.1

normen, 2017-07-06:
It looks like the OpenCV project changed the format of their Haar XML files. I tried to change the converter to import the new file format but although the importer doesn't crash I think I am still making some mistakes, theres always only one stage in the binary file and the OpenMV cam doesn't seem to like the binary files (stalls when detecting).

kwagyeman, 2017-07-06:
We don't support the new OpenCV Haar Cascade file format. Sorry about that. Custom Haar support seems to constantly have issues... Mmm, fixing Haar to support the new file format isn't high on my priority list right now however. Building out support to make DYI robocar racing is what I'm focusing on.

normen, 2017-07-06:
Okay, maybe I'll eventually understand the format and the needed parameters properly to be able to fix the importer... One issue is I don't know where the "threshold" values for each feature are. The stage threshold is a separate value but for the single features I'm not sure if its the last value of the "internalNodes" tuple..?

Anyway is there another way to do what Haar cascades do? You sound a bit dismissive about Haar cascades so mayber theres a different/newer/better way? I want my OpenMV cam to recognize my cat.

kwagyeman, 2017-07-06:
This Has Cascade code was written before I joined the project and I don't have much background on this. Ibrahim's the owner of this code. As far as I understand you have to pass a special flag to OpenCVs Haar generator to make the old Haar file format. The script can't parse the new format.

Let's see what Ibrahim thinks the problem is. He should respond in a few hours.

normen, 2017-07-06:
Cool, thanks. I guess supporting the new format is the best way. I was looking into a converter from new to old but the only way is to actually generate a new cascade file in the old format from test data (as you outlined), which is not what I want to do as there is enough cat cascades out there.

iabdalkader, 2017-07-06:
That's true, you'll need to train the classifier (Not an easy process) and use the old format flag. I agree we should update to the new format at some point, but it's not really top priority. Keep in mind, this feature mainly exists to detect faces and eyes using the built-in cascades.

normen, 2017-07-06:
Well I thought the idea of the OpenMV cam was exactly things like that, detecting objects? Am I missing some obvious other way to do that with OpenMV cam?

iabdalkader, 2017-07-06:
No, the main idea of OpenMV is to be easy to use and support different things, not just object detection. And no, there's no other way to detect cats faces currently that I would recommend. I'll get around to supporting the new format eventually, but please understand that we have limited time and many features and bugs to tend to, so we have to prioritize things. Supporting the new format is Not really easy to do, I'll have to change the binary converter script, the detector and all built-in cascades. Also note the VJ detector was added originally to detect human faces and eyes (cat person speaking btw) and it does that well enough, so cats will have to wait :)

normen, 2017-07-06:
Sorry, I didn't want to come across dismissively. I actually wanted to know the answers to the questions I asked.

As said, I managed to get the converter to work somewhat with new files but as I don't really know what all the parameters are and do theres still issues when the cascade files are actually supposed to be loaded (cam stalls). To me it looks as if all the needed data is still available in the new files.

I'll look into forking the project and making the changes to the script that I have locally now. Maybe somebody more knowledgeable can look into it as well then.

kwagyeman, 2017-07-06:
The reason this hasn't been done is because it's a lot of work. But, I can bump the priority since a lot of folks have been asking about it. This is kind of a good 3 to 4 weeks of work however. So, I've been avoiding doing it and picking off more low hanging fruit.

normen, 2017-07-06:
Okay, I still think you might get around larger changes in the OpenMV code by simply changing the importer script though. I'll send the PR when I'm home and tried around a bit more, maybe it helps.

Thanks for all the info! (And of course for this amazing project!)

kwagyeman, 2017-07-06:
If you can fix it great. Also, if you were to rewrite it in C/C++ I could then integrate it into the IDE.

normen, 2017-07-07:
So heres the changes I made to the importer, only the actual conversion is changed, not info and header generation. Sorry for any confusing code, python isn't really my "native" language :)

https://github.com/normen/openmv/commit/c30f0616cb7a3265da8a957df9a4a71f5182f9dd

The stages count is correct now, the other counts and values seem to match as well but the created cascade still makes the cam "crash" to white/blue blinking after a while.

Theres anyway a few things I don't know:
- I am not really sure how the correlation between the rects and the single features is working with the new format, the numbers don't seem to match and they're in completely separate parts of the xml file now.
- Threshold values should be correct now but that assumption is based on very few info I could find on the file format: http://answers.opencv.org/question/8418/explanation-of-cascadexml-in-a-haar-classifier/

The cascade files I use are the ones I linked in the first post: https://github.com/opencv/opencv/tree/master/data/haarcascades

normen, 2017-07-09:
So after playing around with the OpenMV cam it looks like you're right, its not really made for object recognition. Its very slow with April tags already. I guess I'll resort to using a RasPi Zero with a camera, its not much more expensive but has a GPU :)

Anyway as I still have the cam and development setup I'd still help and get this converter going if you'd give me a few pointers regarding my questions.

kwagyeman, 2017-07-09:
What speed were you looking for with AprilTags?

The zero actually is only 2x faster. With our next system based on the H7 processor we'll be able to  meet or exceed it's performance.

Does OpenCV use the GPU on the zero? For basic filtering ops I can see that. But, doing a Haar Cascade requires a lot of conditional logic.

Anyway, Ibrahim will have to help you out with this. That said, I think I'm going to work on the Haar code next since a lot of folks have been asking about it.

normen, 2017-07-09:
The color blob tracking works at a speed that is suited for my project, i.e. Above 20 fps. I just tried AprilTags to get an idea of how other recognition methods perform.

OpenCV isn't supported on the Broadcom (yet?) but theres a few guys who realized GPU accelerated object recognition on it:
http://www.cnx-software.com/2017/04/30/gpu-accelerated-object-recognition-on-raspberry-pi-3-raspberry-pi-zero/
https://joakimsoderberg.github.io/catcierge/
Until these are released I can use the existing CPU-based methods.

Anyway, this isn't about my project but about the OpenMV cam :) As said, with a bit more information I could go on trying to fix this, otherwise I can help testing any changes. I'll keep my eye on this issue thread.

kwagyeman, 2017-07-09:
Mmm, yeah, lots of code to write in the mean time then. Since we can push firmware updates to all our users we can probably get Haar working with the newer opencv cascade types if some work is put in on that.

To fix this... as far as I can tell, the new and old opencv cascade types are completely different. I've googled for converts on that and haven't seen anything easy. It seems like we'll have to re-write all the Haar code/maintain two version for the new and old opencv types.

Anyway, I'll get to working on general object recong after I've finished working on more of the lower level CV stuff where we have great performance.

normen, 2017-07-09:
Oh, okay. I have no idea about the actual data, to me it looked like most of the data was similar and maybe one or two parameters had to be "prepared" through some of the added data.

kwagyeman, 2017-07-09:
Try saving cascades with the new and old format:

http://docs.opencv.org/2.4/doc/user_guide/ug_traincascade.html

And see the differences.

normen, 2017-07-09:
I don't have the OpenCV SDK installed on my computer and its kind of a pain to do so on a Mac but if you do and already have the chance to create two simple files with the same data that would certainly help.

From the description on that page and the comparison of the cascade files in the OpenMV repo and the OpenCV repo I couldn't see much difference but some more data so I still have no angle here (except UTSL-ing the actual OpenCV code but that'd be quite the detour for me..)

normen, 2017-07-09:
Hmmmmmm... Looking at that importer I linked above they seem to do exactly what I (intended) to do to support the new format. I guess I just made some silly mistake, I will try around a bit more...

normen, 2017-07-10:
Okay, I made a PR, theres still issues, I guess due to the different feature counts.. Funny thing is that the importer I linked above rejects such files with an error.. Hmm..

kwagyeman, 2017-07-10:
Okay, I'm passing the buck for this to Ibrahim, he'll take a look at it. Thanks for contributing! We need more of that! :)

normen, 2017-07-10:
Hah, tell me about it. We have issues finding contributors for the jMonkeyEngine as well, theres no shortage of people nagging about features though ;)

normen, 2017-07-10:
This PR fixes the issue and allows loading new format Haar files: https://github.com/openmv/openmv/pull/254

closed 2017-07-10 by normen