openmv/openmv #253: I2C(4) does not exist

jsbyysheng, 2017-07-06:
### Steps to reproduce
1.change "i2c = I2C(2, I2C.MASTER)" to "i2c = I2C(4, I2C.MASTER)" in example "i2c_control.py"

### Expected behaviour
Tell us what should happen
my board is openmv7. I want to use `I2C(4): (SCL, SDA) = (P7, P8) = (PD12, PD13)` as described in [http://docs.openmv.io/library/pyb.I2C.html](url)
### Actual behaviour
Tell us what happens instead
IDE told me "ValueError:I2C(4) does not exist."
### Describe your setup

**Operating system**:

**OpenMV IDE version:**

**OpenMV firmware version:**

kwagyeman, 2017-07-06:
Mmm, okay, would you be interested in fixing this? It shouldn't be too hard to. Please look into the MP code at this I2C driver and you should be able to enable this feature. Otherwise... Ibrahim will get to fix this feature when he has time.

(We honestly need help fixing stuff...)

iabdalkader, 2017-07-06:
It's just not enabled, newer MP enable I2C4 so it will get "fixed" on its own when we update MP.

jsbyysheng, 2017-07-07:
> It's just not enabled, newer MP enable I2C4 so it will get "fixed" on its own when we update MP.
em....hope to get update as soon as possible because I've used I2C4 in my design. Thank u.

jsbyysheng, 2017-07-07:
> Mmm, okay, would you be interested in fixing this? It shouldn't be too hard to. Please look into the MP code at this I2C driver and you should be able to enable this feature. Otherwise... Ibrahim will get to fix this feature when he has time.

>(We honestly need help fixing stuff...)

I'm sorry that I'm not good at stm32.

kwagyeman, 2017-07-07:
If you look at the latest MP code and compare it to what we have related to I2C support you probably can generate a code diff to apply to fix this issue.

iabdalkader, 2018-04-19:
Enabled.

closed 2018-04-19 by iabdalkader