openmv/openmv #261: Distorted HVGA

cyrtodactylus, 2017-08-28:
### Steps to reproduce
1. Open "helloworld.py"
2. Set framesize as: "sensor.set_framesize(sensor.HVGA)"
3. Preview in IDE (or send compressed JPEG to Matlab via USB)

### Expected behavior
Something that resembles other frame sizes (like this QVGA example)
[image]

### Actual behavior
HVGA image is distorted
[image]

### Describe your setup

I reproduced this on multiple M7 cameras with the following setup:

**Operating system**:
Windows 10

**OpenMV IDE version:**
1.7.1

**OpenMV firmware version:**
2.5.0

kwagyeman, 2017-08-28:
Thanks for finding this bug. I added all the new frame size in mass and I believe only verified everything below QVGA. It's probably a simple bug do to the fact that VGA support is new and uses a Bayer image internally instead of RGB565 for color.

[qvga]<https://user-images.githubusercontent.com/8619467/29784949-b8643ffe-8bda-11e7-8c16-8665240a19de.jpg>

[hvga]<https://user-images.githubusercontent.com/8619467/29784922-9e6390be-8bda-11e7-948b-62a4612cc83f.jpg>

iabdalkader, 2017-08-28:
Hi,
The HVGA resolution, as well as some other resolutions, are invalid. As @kwagyeman mentioned those were Not really tested. The sensor supports the following resolutions: VGA, QVGA and CIF, and it supports downscaling **from CIF** down to 40x30.

I removed all invalid resolutions from the FW. If you need HVGA (480x320) you can use set the resolution to VGA and use `sensor.set_windowing((480, 320))`

kwagyeman, 2017-08-28:
HVGA should be a valid resolution. Why is it invalid?

iabdalkader, 2017-08-28:
> HVGA should be a valid resolution. Why is it invalid?

Can't downscale from VGA. Only from CIF.

kwagyeman, 2017-08-29:
Ah, yeah. I guess that's a camera limitation? Too much work.to<http://work.to>.fix in software to support this res? The reason I ask is because reducing the res does averaging for you of pixels whereas windowing does not.

iabdalkader, 2017-08-29:
@kwagyeman Yes, cam limitations, downscaling only works from CIF down to 40x30 so I also removed the smaller resolutions, and ones which don't have width multiple of 8 for JPEG.

cyrtodactylus, 2017-08-29:
Thanks for looking into this guys. You cleared some things up for me. I was interested in HVGA because I thought it would acquire faster--mistakenly thinking that acquisition rate was a simple function of resolution. The IDE tricked me since QQVGA reports a higher frame rate than QVGA unless the video is disabled.

At any rate, I'd hate to lose the invalid resolutions smaller than QVGA. As @kwagyeman mentioned, they're useful in the absence of a downsampling function and don't appear to decrease frame rate at all compared to QVGA.

closed 2017-08-29 by cyrtodactylus

kwagyeman, 2017-08-29:
The frame rate doubles for anything QVGA and lower. It has to do with how the camera works. It just runs at 120 FPS at QVGA and lower. Only 60 FPS at > QVGA.

Anyway, seems like there's still plenty of smaller resolutions. I'll add them back if I ever have time to modify sensor.c to support them (I will likely never have time to do that though)...