openmv/openmv #262: Periodic FPS drops when using ImageWriter

cyrtodactylus, 2017-09-04:
I've been using the new ImageWriter class and noticed that there are frequent large drops in FPS that are only present when I write image data to SD. Even when I limit FPS to a point well below the largest FPS drops, there are still occasional stutters (as can be seen in the figure below). Any idea why this might be happening?

I was wondering if this might be caused due to the overhead of expanding the file size. I've done a bunch of tests, and sometimes the FPS drops occur at suspiciously regular intervals. Could this be improved with an ImageWriter method that would pre-allocate file size based on a user-defined resolution and frame number? I think that the FatFs _f_lseek_ function can be used to pre-allocate, but I was looking around the OpenMV source code and this seems out of my comfort zone to implement.

[image]

The representative results above were generated with the following code:

```
import sensor, image, time, pyb

sensor.reset()                      # Reset and initialize the sensor.
sensor.set_pixformat(sensor.GRAYSCALE) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)   # Set frame size to QVGA (320x240)
sensor.set_windowing((0, 0, 160, 120))
sensor.skip_frames(time = 2000)     # Wait for settings take effect.
clock = time.clock()                # Create a clock object to track the FPS.

rand_file_num = pyb.rng()
img_writer = image.ImageWriter(str(rand_file_num)+'.bin')

i = 0

fps = 20
mspf = round(1000/fps)

while (i < 1000):
    clock.tick()
    frame_start = pyb.millis()
    img = sensor.snapshot()
    img_writer.add_frame(img)
    frame_end = pyb.millis()
    delta_time = frame_end - frame_start
    if (delta_time < mspf):
        pyb.delay(mspf - delta_time)
    print(clock.fps())
    i = i + 1
    
img_writer.close()
```

### Describe your setup
M7 with a 32 GB, Class 10, SDHC Sandisk (similar results using 4 GB Class 4--just with a lower average FPS)

**Operating system**:
Windows 10

**OpenMV IDE version:**
1.7.1

**OpenMV firmware version:**
2.5.0

kwagyeman, 2017-09-04:
Mmm, I'm working on the code right now. I can whip something up for you by tomorrow. However, the problem is basically do to the SD card stalling the OpenMV Cam when doing a write. At the level we talk to the SD card... we just do a multi-block write. So, basically the image is pushed in one large write call to the lower level FS. As much buffering is done as possible in the OpenMV Cam firmware to get the image data setup to be streamed in 512 byte chunks quickly. However, I can't do anything about the SD card needing time to erase a flash page. As for file seeks, those usually aren't a problem in my experience. Read time is almost always blazing fast.

I'm not sure how to solve this issue actually. If the camera were made for video recording you'd normally implement a fifo buffer to store images when this happens. However, we don't have the RAM to buffer images. So, if we get blocked by the SD card we have to just drop the next frame.

Alternatively, pre-erasing the SD card could be done to free up blocks. And... by erase I mean issuing commands to get the SD card to flash erase all it's pages. However, the STM32 HAL and FAT16/32 in general won't let us work at that level.

I can try the file pre-allocation if you'd like, this is easy to add. Not sure it will matter. Let me know.

cyrtodactylus, 2017-09-04:
I see...looks like you've thought about this quite a bit. I'd gladly try
out the pre-allocation thing, but only if it's as easy as you say. I don't
want to waste your time.

<https://github.com/openmv/openmv/issues/262#issuecomment-326844781>, or mute
the thread

kwagyeman, 2017-09-04:
Can you compile code? If so, I'd prefer to enable you to tackle problems yourself and send us PRs to fix things. Otherwise I'll build you a firmware test image. All you have to edit is right here:

https://github.com/openmv/openmv/blob/master/src/omv/py/py_image.c#L4051
https://github.com/openmv/openmv/blob/master/src/omv/py/py_image.c#L4026

So, after the file is created (file_write_open) just do the f_tell call to preallocate and then before close do the truncate call to remove unused bytes. No file buffering is used in this code so you don't have to worry about anything being cached in RAM. All writes are sent directly to the SD card.

cyrtodactylus, 2017-09-04:
Yeah, I can give it a shot. I'll install the development environment sometime over the next couple of days and make those changes. I'll let you know how it goes

Thanks

cyrtodactylus, 2017-09-05:
Unfortunately I think you were very right about pre-allocation. It didn't seem to help much. In all my tests the standard deviation dropped a bit (3.5386 to 2.7100 in the example below), but qualitatively I think they both look about the same
[image]

Do you think that transferring images via SPI to a second microcontroller (with more RAM) and saving to SD there would be better, worse, or about the same? I stream data created from machine vision analyses via USB to a PC for real-time use, but this is all for a scientific application and I need the video data just in case any of the real-time results need to be validated. However, I don't care what happens to the video as long as it's eventually saved and doesn't interfere too much with the real-time data stream...

kwagyeman, 2017-09-05:
Mmm, here's my though process on this:

The current architecture of the OpenMV Cam doesn't have any fifo buffers between the camera data stream and the sd card output stream. This makes it simple to use since each new frame you get is basically the latest image and you don't have to worry about processing old stuff if you fall behind. For high speed movement this is actually a must. However, if there was an image fifo buffer you wouldn't have to worry about frames being dropped and the FPS in general would increase because the processor would be doing more work instead of waiting for the next frame. But... for large resolutions... we don't have the RAM to store multiple images.  If we did have the RAM then you'd move frame capturing to an interrupt and store the data in an unused buffer... but, I don't see us ever doing that on any embedded platform because we use extra ram for CV stuff when we can.

Okay, so, could a fifo buffer be put in front of the SD card? I suppose the lack of RAM issue prevents this too. Given raw images aren't processed there's basically nothing happening except camera data being immediately written to the disk.

I am concerned the the write speed of the disk seems too low a bit. I did some performance tests and found we're only getting about 2.5MB/s out of 12.5MB/s we could get (minus overhead). Read is blazing fast though... I've hit 300 FPS with that when I removed the frame limiter.

...

Okay, so I just ran an 100 second test with my cam and I never dropped before 30 FPS with the FB disabled. That said, I saw the frame rate go up and down. It was not perfectly stable. The average was about 40 FPS for RGB565 QQVGA.

So, I guess the SD card write pauses are injecting glitches here.

...

Okay, so, I don't think there's much that can be done onboard the camera without tossing the whole architecture and making the camera a fixed function device by using all the RAM as fifos.

If you instead JPG compress the image data and send it via SPI to another microcontroller you won't have to worry about large frame rate dips (note I said JPG compress... because RAW images are huge and SPI can't handle them). To write a jpg image just do: spi.write(img.compress()). And... you have to control the CS wire yourself. Make the OpenMV Cam the master and the thing that's logging the data a slave.

That said, is there a reason your application can't handle the occasional dip? The average is still correct. Can't you interpolate between results to provide a smooth output to whatever is receiving said results?

cyrtodactylus, 2017-09-05:
Yeah, the frame drops aren't ideal, but I think 40 fps average should be tolerable for our purposes. Thanks for looking into all of this.

One last question: What's the make and model of your SD card? You never dropped below 30 FPS during your test, but my card dropped below 30 multiple times. I was reading around on some forums and people were showing substantial differences in the maximum busy delays measured in different cards during writing. I imagine modern cards are not optimized for single block writes, but maybe some are better than others.

kwagyeman, 2017-09-05:
I'm just using a cheap 32 gb SD card I got off of Amazon. It says San disk on it and it's black. Note that the OpenMV Cam is doing multiblock writes. To get good performance we write the whole image in one multi block write operation. So, this would be 19 KB or so for a 160x120 image.

iabdalkader, 2017-09-05:
@cyrtodactylus I'm using a UHS card, I'll test and report back.

iabdalkader, 2017-09-05:
Here you go, is this better ?
[image]

cyrtodactylus, 2017-09-05:
It's hard to say. Doesn't look so bad, but there can be a lot of variability between different 1000 frame epochs. I'd guess that you just got lucky. Even in my test of 10000 frames there were only 3-4 times where fps dropped below 20 fps. It's not ideal, but I think I can live with it.

Thank you both for thinking about this issue.

closed 2017-09-17 by kwagyeman