openmv/openmv #265: USB device descriptor request failed based on OpenMV2

orta1986, 2017-09-22:
Hello, I designed a new board based on Openmv2 schematic with STM32F427VIT6 ， and I faced a odd problem.

When my board power on with BOOT1 = 0  BOOT0 = 1,   DFU mode is correctly recognized and openmv.dfu (from openmv/firmware/OPENMV2/) is updated successfully.

Then I turn BOOT0 = 0  and  replug USB,  "USB device descriptor request failed" is shown in Windows . 

I have checked Osillator , BOOT0 BOOT1 pin and 3.3V, all are correct.

Therefore I am confused where problem might be.

Anyone can help me please........

Thanks .

kwagyeman, 2017-09-22:
I can't really say what's wrong. You need to narrow the issue down to a particular questions. Since you changed things you could have shorted something, used a different crystal, etc. I guess, to start the debugging process. You've verified that DFU works. So, the camera should have blinking blue light when power is applied after you've programmed it via DFU. Is this correct?

orta1986, 2017-09-22:
Hello Kwabena,

Thank you for your reply. 
My design is similar as the openmv2 sch, but I just solder  STM32 chip ,  camera and another  peripheral are not soldered. So we can focus on the chip to debug.

After the dfu downloaded and reboot STM32, I found that a LED start blink (PC2 pin => LED_G). Because I have not soldered another two LEDs, I am not sure whether the two LEDs blink . I will test it later.

Therefore, according to  my describe, can we infer that STM32 works correctly?

My crystal is 8Mhz + 22pF.

Thank you so much.
Best regards,

Lizhong

 

Sent from YoMail for Gmail

kwagyeman, 2017-09-22:
If everything is working right the led led on the board should be blinking. Or, whatever you have attached to the blue LED pin. Note that only the blue LED should be blinking. Any other led blinking is bad.

Finally, please use a slightly older version of the OpenMV Firmware. Before the frame rate increase for your M4 device. We found out the frame rate increase didn't work on all M4s.

orta1986, 2017-09-22:
Hello kwagyeman,
Thank you for your help and I have found what the problem is.  The crystal I used is 8Mhz , but it should be 12Mhz. I change the crystal and the usb is recoginzed corretly( stm32 is recoginzed as a disk).

But...I faced another issue.
When I open Openmv IDE and clikc "Connect", the Connect Logo changed and the board information is shown below (Fireform Version 2.5.0, Serial Port : COM11 , Drive : F:/). 
Then I open Example -> 01 Basic -> main.py ， and click "start".   There is no reaction on IDE , the Start logo does not change to a "red X". I also tried to click start several times but it still not run. 

Could you please give me a cue?

Thanks

kwagyeman, 2017-09-22:
When you tell a script to run via OpenMV IDE it just sends the script to the camera and then polls the cameras script execution state. There's not an interlock or anything like that. Anyway, so, you seem to be almost there. I think the next thing is to try out an example script that just prints text or blinks a light. If those work then try out the camera.

orta1986, 2017-09-22:
this is my script:

import time, pyb
led = pyb.LED(3) 
var = 1
while(var == 1):
    led.on()
    time.sleep(1500)
    led.off()
    time.sleep(1000)
    led.on()
    time.sleep(1500)
    led.off()
    time.sleep(1500)

it is a simple led blink , but it does not work...

orta1986, 2017-09-22:
The board use is recognized as " OpenMV Cam USB COM Port (COM11)", is it right?

kwagyeman, 2017-09-22:
Yeah, if connect didn't give you problems then you are mostly there. Connecting to the camera executes a ton of code.

Um, try reflashing the camera using OpenMV IDE. You can do this by clicking on the firmware version label which is actually a button and following the prompts.

orta1986, 2017-09-22:
OK, I have reflashed the board by click on "Fireware Version" and 2.5.0 is still a latest version. 
I just found a situation that when I click "start" (no reaction) and click it again or more,  a message box is shown as " Busy ...Please wait... " . If we wait for seconds(about 20-30 sec) , the usb connection is disconnected.

So, is it problem about the communication of "Start script" ? Feel like a stuck.

orta1986, 2017-09-22:
Where can I download old version of openmv2 dfu?  2.0 2.4 etc. I worried if the latest dfu can not support all chip of stm32? My chip is STM32F427VIT6

kwagyeman, 2017-09-22:
On the OpenMV GitHub we have old releases of the firmware. You can download old firmware from there.

As for the error you are getting... That means the IDE never got a response from the camera for a command.

Can you try using the IDE on another PC?

orta1986, 2017-09-23:
hello ， I have tested it on my PC(windows10) and Macbook OS , both connection of IDE on systems are failed. 
So far, I have no idea what problem is, I start to doubt the stm32 chip?  Maybe I have to buy different chip (STM32F765 + openmv3 or STM32F407+openmv1) and redesign PCB board. It will take me more time to debug it...

orta1986, 2017-09-23:
is it possible that the openmv2 fireware can not support STM32F407VIT6 ? 
I found that the official board is based on STM32F407VGT6, but my board is STM32F407VIT6 . Maybe the fireware DFU (or IDE) can not support the pid & vid of STM32F407VIT6 ?

kwagyeman, 2017-09-23:
We support the STM32F427VGT... Note sure where you got the 04 from.

orta1986, 2017-09-23:
Sorry I make a mistake in writing. I will repost as:

I have tested it on my PC(windows10) and Macbook OS , both connection of IDE on systems are failed.
So far, I have no idea what problem is, I start to doubt the stm32 chip? Maybe I have to buy different chip (STM32F765 + openmv3 or STM32F407+openmv1) and redesign PCB board. It will take me more time to debug it...

is it possible that the openmv2 fireware can not support STM32F427VIT6 ?
I found that the official board is based on STM32F427VGT6, but my board is STM32F427VIT6 . Maybe the fireware DFU (or IDE) can not support the pid & vid of STM32F427VIT6 ?

kwagyeman, 2017-09-23:
Um, we set the PID and VID in software. Anyway, yeah, if you are using the VI processor the flash layout changed... which will break the uPY system and possibly corrupt the main firmware image. You should use the exact same hardware build.

orta1986, 2017-09-23:
OK I have bought a STM32F427VGT6 , and I will test it tomorrow. Thank you so much.

iabdalkader, 2018-04-26:
If you need more help with your project, please post on the forums project discussion section.

closed 2018-04-26 by iabdalkader