openmv/openmv #266: Is there any problem with the latest makefile ?

Fjinwei, 2017-09-29:
### Expected behaviour
When I make in /openmv/src,it should generate modstm_qstr.h after 

> Create /mnt/hgfs/ShareWithUbantu/openmv/src/build/micropython/genhdr/pins.h

### Actual behaviour
But there is something bad happened.

> mkdir -p /mnt/hgfs/ShareWithUbantu/openmv/src/build/micropython/genhdr
Create /mnt/hgfs/ShareWithUbantu/openmv/src/build/micropython/genhdr/pins.h
make[1]: *** No rule to make target '/mnt/hgfs/ShareWithUbantu/openmv/src/build/micropython/modstm_qstr.h', needed by '/mnt/hgfs/ShareWithUbantu/openmv/src/build/micropython/genhdr/qstrdefs.generated.h'.  Stop.

Maybe I should not think about there is something wrong with the latest makefile ,but I just don't know how to deal with this.
### Describe your setup

**Operating system**:ubantu16 in VMware

kwagyeman, 2017-09-29:
Ibrahim is the only one who knows about this and he is on vacation right now. Um, question, what are you trying to do? The current make file works... So, what's the issue exactly?

Fjinwei, 2017-09-30:
Just compile the firmware.When i make with TARGET=OPENMV3,everything is almost alright.But with TARGET=OPENMV1 or OPENMV2,I meet the problem described bdfore.

Fjinwei, 2017-09-30:
Just compile the firmware.When i make with TARGET=OPENMV3,everything was almost alright.But with TARGET=OPENMV2 or OPENMV1,i met the problem described before.

kwagyeman, 2017-09-30:
Ibrahim mentioned to me that the OpenMV Cam 2 build was broken when updating to MP 1.9.2. He will fix it when he gets back from vacation.

iabdalkader, 2017-09-30:
The M4 cam hasn't been tested since the last update but I'm working on it now.

iabdalkader, 2017-10-03:
@Fjinwei The M4 build should be working now.

closed 2017-10-12 by iabdalkader