openmv/openmv #268: blob.density returns 0.00 all the time

hansondon, 2017-10-04:
### Steps to reproduce
1. Find blob in an image
2. Call blob.density()

### Expected behaviour
The return value should be a legit float number <= 1.0

### Actual behaviour
Instead, only 0.000 is returned

### Describe your setup

**Operating system**: Ubuntu 16.04

**OpenMV IDE version:** 1.7.1

**OpenMV firmware version:** 2.5.0

### My hypothesis
It seems the implementation of this method simply divides blob.pixels() by blob.area(), which is an integer division and will only yield 0 because the result is smaller than 1.
### My workaround
Since it is impossible to import from __future__, I can create my own density() method where I do the following instead:
return float(b.pixels())/b.area()

But I do hope this can be fixed.

kwagyeman, 2017-10-04:
Thanks for catching this. I added this as a convenience method but I guess messed up there. It will be fixed.

kwagyeman, 2017-10-04:
Wait, can you submit a PR to fix? That would be better.

iabdalkader, 2017-10-04:
@kwagyeman Already fixed it.

closed 2017-10-11 by hansondon