openmv/openmv #272: set_windowing doesn't work

kwagyeman, 2017-10-14:
### Steps to reproduce
sensor.set_pixformat(sensor.QQVGA)
sensor.set_windowing((0, 80, 160, 80))

### Expected behaviour
You should get a 160x80 image

### Actual behaviour
You get that plus a lot of garbage. I'm guess the frame buffer size is not getting setup right and the DCMI hardware works... but, the previous (QVGA) image size gets set to the PC.

### Describe your setup

**Operating system**:
Windows 10

**OpenMV IDE version:**
v1.7.1

**OpenMV firmware version:**
v2.5.0

iabdalkader, 2017-10-14:
The y offset is set to 80 and the height is also set to 80. Use (0, 80, 160, 40) or (0, 40, 160, 80)

iabdalkader, 2018-04-25:
This is not fixed (can't fix it) but now I check the window bounds before calling set_window.

closed 2018-09-20 by kwagyeman