openmv/openmv #274: Master not compiling (-Werror unused-functions)

AlexisTM, 2017-10-16:
The current master does not build as 3 functions are not used. 

It can be *solved* by removing -Werror from the Makefile

```
img/lsd.c:810:18: error: 'new_image_int_ini' defined but not used [-Werror=unused-function]
 static image_int new_image_int_ini( unsigned int xsize, unsigned int ysize,
                  ^
img/lsd.c:876:21: error: 'new_image_double_ptr' defined but not used [-Werror=unused-function]
 static image_double new_image_double_ptr( unsigned int xsize,
                     ^
img/lsd.c:974:21: error: 'gaussian_sampler' defined but not used [-Werror=unused-function]
 static image_double gaussian_sampler( image_double in, double scale,
                     ^
```

iabdalkader, 2017-10-16:
What's your GCC version and the build target ?

kwagyeman, 2017-10-16:
#pragma GCC diagnostic ignored "-Wunused-function"
#pragma GCC diagnostic ignored "-Wunused-variable"

Is in that code... you're GCC version is ignoring them?

AlexisTM, 2017-10-16:
My gcc version is 5.4.1; 

> $ arm-none-eabi-gcc -v
> Using built-in specs.
> COLLECT_GCC=arm-none-eabi-gcc
> COLLECT_LTO_WRAPPER=/home/alexis/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/lto-wrapper
> Target: arm-none-eabi
> Configured with: /home/build/work/GCC-5-0-build/src/gcc/configure --target=arm-none-eabi --prefix=/home/build/work/GCC-5-0-build/install-native --libexecdir=/home/build/work/GCC-5-0-build/install-native/lib --infodir=/home/build/work/GCC-5-0-build/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/home/build/work/GCC-5-0-build/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/home/build/work/GCC-5-0-build/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/home/build/work/GCC-5-0-build/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/home/build/work/GCC-5-0-build/install-native/arm-none-eabi --build=i686-linux-gnu --host=i686-linux-gnu --with-gmp=/home/build/work/GCC-5-0-build/build-native/host-libs/usr --with-mpfr=/home/build/work/GCC-5-0-build/build-native/host-libs/usr --with-mpc=/home/build/work/GCC-5-0-build/build-native/host-libs/usr --with-isl=/home/build/work/GCC-5-0-build/build-native/host-libs/usr --with-cloog=/home/build/work/GCC-5-0-build/build-native/host-libs/usr --with-libelf=/home/build/work/GCC-5-0-build/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for ARM Embedded Processors' --with-multilib-list=armv6-m,armv7-m,armv7e-m,armv7-r,armv8-m.base,armv8-m.main
> Thread model: single
> gcc version 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715] (GNU Tools for ARM Embedded Processors)

iabdalkader, 2017-10-16:
I'm using GCC version 6.2.1. doesn't show those warnings.

iabdalkader, 2018-04-25:
This is outdated by now.

closed 2018-04-25 by iabdalkader