openmv/openmv #275: CRC solution proposed is not working

AlexisTM, 2017-10-16:
Trying to use the CRC solution to overcome the micropython integer restriction fails.

http://docs.openmv.io/reference/asm_thumb2_hints_tips.html?highlight=crc#overcoming-micropython-s-integer-restriction

With the code:
```python
from array import array
import stm

def enable_crc():
    stm.mem32[stm.RCC + stm.RCC_AHB1ENR] |= 0x1000

def reset_crc():
    stm.mem32[stm.CRC+stm.CRC_CR] = 1

@micropython.asm_thumb
def getval(r0, r1):
    movwt(r3, stm.CRC + stm.CRC_DR)
    str(r1, [r3, 0])
    ldr(r2, [r3, 0])
    str(r2, [r0, 0])

def getcrc(value):
    a = array('i', [0])
    getval(a, value)
    return a[0] & 0xffffffff # coerce to arbitrary precision

enable_crc()
reset_crc()
for x in range(20):
    print(hex(getcrc(0)))
```

We got the error:

> SyntaxError : 'movwt' expects an integer

iabdalkader, 2017-10-16:
This works with MicroPython v1.9.2.

iabdalkader, 2017-10-16:
Note for future issues, please use the issue template.

closed 2017-10-16 by iabdalkader