openmv/openmv #313: USB_VCP send errors

cyrtodactylus, 2018-02-12:
I'm sending strings of ~215 characters to Matlab each frame using the USB_VCP class. I've noticed a low percentage of errors in the form of character mutations. I quantified this by sending a known string (213 characters long) 100000 times and 205 (0.2%) of those strings had at least one character that was different from the target string.

Are these numbers of errors expected? I thought there was a certain amount of error correction for each packet built into USB

### Describe your setup

**IDE version:**
1.8.0

**Firmware version:**
M7, 2.6.0

**Operating system:**
Windows 10

iabdalkader, 2018-02-12:
Can you post the Python script ?

kwagyeman, 2018-02-12:
Hi, are characters being dropped? The VCP class has no flow control. So you can't send it characters too fast otherwise it drops characters. What's your frame rate? 215 characters doesn't seem that high per frame at 30 FPS. You'd only have issues I'm thinking if you're printing 10s of thousands of characters per frame.

cyrtodactylus, 2018-02-13:
This is the script that I used to run my test. Basically I send a character from matlab to start "acquisition mode" and then let it run until it sends 100000 strings. It's a simplified version of my real code where I have additional "modes" and the string is generated using information from find_blobs.

Thanks,
David

```
import pyb, sensor, time, image, ustruct

red_led   = pyb.LED(1)
green_led = pyb.LED(2)
blue_led  = pyb.LED(3)

usb = pyb.USB_VCP()
usb.setinterrupt(-1)

sensor.reset()                              # Reset and initialize the sensor.
sensor.set_pixformat(sensor.GRAYSCALE)      # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)           # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)             # Wait for settings take effect.
sensor.set_auto_gain(False)                 # must be turned off for color tracking
sensor.set_auto_whitebal(False)             # must be turned off for color tracking

# Helper functions

def blue_blink():
    blue_led.on()
    time.sleep(150)
    blue_led.off()
    time.sleep(100)
    blue_led.on()
    time.sleep(150)
    blue_led.off()
    time.sleep(600)

def red_blink():
    red_led.on()
    time.sleep(150)
    red_led.off()
    time.sleep(100)
    red_led.on()
    time.sleep(150)
    red_led.off()
    time.sleep(600)

def green_blink():
    green_led.on()
    time.sleep(150)
    green_led.off()
    time.sleep(100)
    green_led.on()
    time.sleep(150)
    green_led.off()
    time.sleep(600)

def leds_off():
    red_led.off()
    green_led.off()
    blue_led.off()

def leds_on():
    red_led.on()
    green_led.on()
    blue_led.on()

while(True):

    while(not usb.isconnected()):
        mode = -1
        blue_blink()
    
    while(usb.isconnected()):
    
        leds_off()
        if (mode == -1):
            mode = 0
    
        # listen for inputs
        if (usb.any()):
            usbinput = usb.readline()
            if (usbinput == b'A' and mode == 0):
                mode = 2
                green_led.on()
            elif (usbinput == b'a'):
                mode = 0
                green_led.off()
            else:
                leds_on()

        # Acquisition mode
        if (mode == 2):
            img = sensor.snapshot()
            
            frame_string = 'Pupil:1[{x:142, y:96, w:82, h:66, pixels:4299, cx:182, cy:129, rotation:0.075185, code:1, count:1}]Reflection:1,89399,42,76800[{x:187, y:90, w:5, h:5, pixels:28, cx:189, cy:92, rotation:0.959784, code:1, count:1}]' + '\n'
            usb.send(bytes(frame_string, 'ascii'))
```

cyrtodactylus, 2018-02-13:
@kwagyeman whoops, didn't see your post. Frame rate is ~42. I've never seen a dropped character out of the dozens that I've looked at. It's always a character change.

kwagyeman, 2018-02-13:
From looking at the VCP code it has a 512 byte buffer and waits for free space up to a timeout before sending. The buffer should be serviced by the PC multiple times per millisecond.

I'll give the code a test.

kwagyeman, 2018-02-13:
Hi, I can confirm the problem happens even at a low rate of 10 FPS... I see the random character change.

After looking at the USB code... I'm not sure what can be done to fix this? I don't really know where the bug is. I don't think it's on the PC side.

Mmm, so one thing I notice that looks a little... tricky... is that the MP HAL code use the output 512 byte TX fifo to sent data via USB from. It doesn't copy the data into a temp buffer. Wondering if there's an interrupt situation which can cause it to miss fire and corrupt chars.

kwagyeman, 2018-02-13:
Do you have a micropython board? Ibrahim has one. Can someone see if this problem has on a MicroPython board? You should be able to test the issue using the open terminal feature in OpenMV IDE. Just save a script to the board and then view the serial data with the open terminal feature. You should see a random byte change on the scrolling text which your eye will catch easily.

kwagyeman, 2018-02-13:
In the mean time. If you switch to a polling solution from the PC the problems should go away. I.e only send data if the PC asks for it. As in per time you get an "A" send the string giving frame processing. Otherwise don't send any data. So, you just check to see if there's data in the VCP per frame and send a string to the PC after finishing all your processing for the current frame.

cyrtodactylus, 2018-02-13:
I don't have a micropython board.

I tried the polling approach, and I'm still getting character changes (same percentage of corruptions even). Here's what I changed in the python script. Once frame data is collected, it waits for a character from PC before sending the string. Is this what you had in mind?

```
while(True):

    while(not usb.isconnected()):
        mode = -1
        blue_blink()
    
    while(usb.isconnected()):
    
        leds_off()
        if (mode == -1):
            mode = 0
    
        # listen for inputs
        if (usb.any()):
            usbinput = usb.readline()

            if (usbinput == b'A' and mode == 0):
                mode = 2
                green_led.on()
            elif (usbinput == b'a'):
                mode = 0
                green_led.off()
            elif (usbinput == b'f' and mode==3): # PC request for frame data
                mode = 4
            else:
                leds_on()

        # collect frame data
        if (mode == 2):
            img = sensor.snapshot()
            
            frame_string = 'Pupil:1[{x:142, y:96, w:82, h:66, pixels:4299, cx:182, cy:129, rotation:0.075185, code:1, count:1}]Reflection:1,89399,42,76800[{x:187, y:90, w:5, h:5, pixels:28, cx:189, cy:92, rotation:0.959784, code:1, count:1}]' + '\n'
            # usb.send(bytes(frame_string, 'ascii'))
            mode = 3
            
        # send frame data
        if (mode == 4):
            usb.send(bytes(frame_string, 'ascii'))
            
            mode = 2
```

kwagyeman, 2018-02-14:
Ibrahim will have to debug and fix this. I don't have the MP board to check if it's our bug or theirs.

In the meantime, add a checksum to the string and toss it if the checksum fails. The error rate seems to enough to tolerate this. You can do a check sum by adding each byte of a character and sending that value at the end.

iabdalkader, 2018-02-15:
Hi, can you comment snapshot and test again ?

cyrtodactylus, 2018-02-15:
After commenting out snapshot I got no errors from 10000 iterations.

However, my input buffer was filling, so I put in in pyb.delay(5) to replace snapshot. This caused something that I don't understand. I'm getting character errors, but these errors are not randomly scattered throughout the test. For example, during one test I got ~40 errors, but they were only between iterations 2400 and 2900. Another time I got ~30 errors between iterations 9600 and 10000 and the 31st error was on iteration 3104

iabdalkader, 2018-02-15:
Never mind, I think I see the problem.

iabdalkader, 2018-02-15:
Can you test this firmware image ?
https://github.com/openmv/openmv/blob/master/firmware/OPENMV3/firmware.bin

cyrtodactylus, 2018-02-16:
Sweet, I think that worked! I haven't seen any character errors in my tests. I'll follow up if I notice the issue again, but I'm optimistic that this solved the problem.

Thank you both for your help!

closed 2018-02-16 by iabdalkader