openmv/openmv #319: Poll Firmware Version in Firmware

ralora, 2018-02-27:
### Steps to reproduce
1. There is no feature in python to poll for firmware version.
Question was first brought up in forum: http://forums.openmv.io/viewtopic.php?f=3&t=626

### Expected behaviour
A poll should return something like "2.7.0".

### Actual behaviour
Not implemented

### Describe your setup

**IDE version:**
N/A
**Firmware version:**
v2.7.0
**Operating system:**
N/A
### For developers

**GCC version:**
N/A
**Build target:**
N/A

kwagyeman, 2018-02-28:
I'll have to add a new module to hold this info to do it properly. I guess I'll call the module "omv" and it will hold settings and info. I can put something in place over the weekend. Busy working on the imglib right now.

kwagyeman, 2018-04-08:
I've added a new omv module for version info. It will be int he next release.

closed 2018-04-08 by kwagyeman