openmv/openmv #323: USB_VCP `isconnected` function always return false, when connected with OpenMV IDE

mushroom-x, 2018-03-15:
### Steps to reproduce
run this script below 
(function same as  exmaples-> basic->main.py
connect  openmv to the ide, run this script
```python
# Main Module Example
#
# When your OpenMV Cam is disconnected from your computer it will either run the
# main.py script on the SD card (if attached) or the main.py script on
# your OpenMV Cam's internal flash drive.

import time, pyb

# 定义蓝灯
led = pyb.LED(3) # Red LED = 1, Green LED = 2, Blue LED = 3, IR LEDs = 4.
usb = pyb.USB_VCP() # This is a serial port object that allows you to
# communciate with your computer. While it is not open the code below runs.

def blink_led(led):
    led.on()
    time.sleep(150)
    led.off()
    time.sleep(100)
    led.on()
    time.sleep(150)
    led.off()
    time.sleep(600)

while(not usb.isconnected()):
    print('not connected')
    blink_led(led)

# 使用绿灯
led = pyb.LED(2) # Switch to using the green LED.
while(usb.isconnected()):
    print('connected')
    blink_led(led)
```

### Expected behaviour
connect with usb ,  green led blink, and print "connected" in serial.

### Actual behaviour

when I connected openmv with ide , but I always get "not  connect" info in terminal.
write the code in flash (main.py) disconnect pc , also show the "not connect" status(blue  led blink)

### Describe your setup

**IDE version:**
2.8.1
**Firmware version:**
2.7.0
**Operating system:**
Win 10

kwagyeman, 2018-03-15:
You can't use the VCP port when OpenMV IDE is connected because it's using the VCP  port. If you want to use the VCP port for a final application please develop using the standard serial UART and then switch to the VCP once you've fixed all your bugs.

iabdalkader, 2018-03-15:
> write the code in flash (main.py) disconnect pc , also show the "not connect" status(blue led blink)

Note `usb.isconnected()` returns True when CDC is connected (not when USB cable is connected).

mushroom-x, 2018-03-16:
get it , thks.

closed 2018-03-16 by mushroom-x