openmv/openmv #338: micropython neopixel module not working in openmv

dhunjoshi, 2018-04-17:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

DeqingSun, 2018-07-03:
I got it kind of working by writing own inline assemly code. It seems openMV can not pass an array to assembly, so it can only display white for now.

[image]

```
import utime
import pyb
from pyb import Pin

# Connect 8 neoPixels on P9

pin1 = pyb.Pin('P9', Pin.OUT_PP, Pin.PULL_NONE) #using P9, PD14

@micropython.asm_thumb
def neoPixelOut(r0,r1):
    movwt(r5, stm.GPIOD)
    movw(r6, 1 << 14)

    label(loopByte)

    mov(r2, r0) #load a byte

    movw(r3, 8) #iterate Bit

    label(loop_bit)

    movwt(r5, stm.GPIOD)
    strh(r6, [r5, stm.GPIO_BSRRL])  #P9 High

    movw(r5, 1 << 7)    #high level duration
    movwt(r4, 93)   #bit is 1
    tst(r2, r5) #test bit 7 of R2
    it(eq)
    movwt(r4, 31)  #bit is 0

    cmp(r3 ,8)   #compensate for loop byte
    it(eq)
    sub(r4, 6)

    label(delay_high)
    sub(r4, r4, 1)
    cmp(r4, 0)
    bgt(delay_high)
    movwt(r5, stm.GPIOD)
    strh(r6, [r5, stm.GPIO_BSRRH])  #P9 Low

    movw(r5, 1 << 7)    #low level duration
    movwt(r4, 42)   #bit is 1
    tst(r2, r5) #test bit 7 of R2
    it(eq)
    movwt(r4, 118)  #bit is 0

    cmp(r3 ,8)   #compensate for loop byte
    it(eq)
    sub(r4, 12)

    label(delay_low)
    sub(r4, r4, 1)
    cmp(r4, 0)
    bgt(delay_low)

    movw(r5, 1)
    lsl(r2, r5) #shift left for 1 bit

    sub(r3, r3, 1)
    cmp(r3, 0)
    bgt(loop_bit)

    sub(r1, r1, 1)
    cmp(r1, 0)
    bgt(loopByte)

while(True):
    try:
        state = pyb.disable_irq()
        neoPixelOut(5,24)
    finally:
        pyb.enable_irq(state)
    utime.sleep_us(50)   #make sure there is enough time to latch

```

iabdalkader, 2018-07-03:
Enabled uctypes.

closed 2018-07-03 by iabdalkader

kwagyeman, 2018-07-03:
WOW, that's cool. Amazing ASM work.