openmv/openmv #339: sinh not found in math library

aakoch, 2018-04-19:
[Per request on forums](http://forums.openmv.io/viewtopic.php?f=6&t=692), I am opening a ticket.

Math function "sinh" is not supported in math library.

### Steps to reproduce
1. Create file with this code:
```
import math
math.sinh(0)
```
2. Run

### Expected behaviour
No error should be produced

### Actual behaviour
Error is thrown:
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
AttributeError: 'module' object has no attribute 'sinh'
MicroPython v1.9.2-4442-g8d787e97 on 2018-01-24; OPENMV3 with STM32F765
```

### Describe your setup

**IDE version:**
1.8.2

**Firmware version:**
2.7.0

**Operating system:**
macOS High Sierra

kwagyeman, 2018-04-19:
Ibrahim, can you enable this quickly under the MP enable scripts? It should be in there already...

iabdalkader, 2018-04-19:
Fixed

closed 2018-04-19 by iabdalkader