openmv/openmv #348: No module named 'ussl'

martindr, 2018-05-23:
### Steps to reproduce
1. In the OpenMV IDE start a new file.
2. import ussl 
3. Connect to OpenMV 7 and run the program

### Expected behaviour
Program should compile and run

### Actual behaviour
Error: no module named 'ussl'

### Describe your setup
**IDE Version** 
1.9.0

**Firmware version:**
2.9.0

**Operating system:**
Windows 10

iabdalkader, 2018-05-24:
This module is not implemented for Micropython stm32 port.

Ascarron, 2019-07-18:
Sorry for re-opening this issue. The module was recently updated and opened for stm32 port. Could you please include it in the next firmware build?

kwagyeman, 2019-07-18:
Yes, will do.

dipeshwor, 2019-09-04:
Hi, I am getting a '301 permanently moved' error when accessing any https link. The usocket in OpenMV works for http links like the open weather data API but gives the 301 error for https links like Canvas API. I came across this issue in the forum and think it's related to what I am trying to do. I probably need the ussl module for the usocket to work properly. Please let me know if there are any updates on the issue. Thanks.

iabdalkader, 2019-09-04:
It's on my TODO list for WINC, I'll also update the module firmware and driver.

closed 2019-10-25 by iabdalkader

iabdalkader, 2019-10-26:
Hi, the ussl module is now available in the [latest beta release](https://github.com/openmv/openmv/releases/tag/v3.5.0.beta.3) (with mbedtls), additionally the WINC1500 host driver and firmware have been updated to the latest (v19.3.0/v19.6.1). Note the new release requires a WINC1500 firmware update (included with the release).

PR #623 (2019-10-25): Wifi updates
Fix recv() bug introduced after WINC1500 host driver/firmware updates.
Enable modussl via mbedtls.
Update HTTP client example.
Add HTTPS client example.
Fixes #348