openmv/openmv #369: pyb.USB_VCP().isconnected() is not work in windows

kidswong999, 2018-08-24:
### Actual behaviour

isconnected() should return true, but it return false in Windows.
It works fine on MacOS

**test code:**

```
import pyb, time
while True:
    print(pyb.USB_VCP().isconnected())
    time.sleep(1000)
```

**IDE version:**
2.2.0
**Firmware version:**
3.3.0
**Operating system:**
Windows 10

kwagyeman, 2018-08-24:
Ibrahim will debug this on Windows.

iabdalkader, 2018-08-24:
I know the issue, seems Windows 10 serial driver doesn't set DTR (MP depends on it to detect CDC open port). Now I'm looking for a workaround, something to use with DTR (maybe set line coding) but it may not work (so maybe there isn't a fix). I'll know more once I have the USB analyzer capture.

iabdalkader, 2018-09-07:
@kidswong999 This issue has been fixed in the IDE. If you want to test the pre-release IDE, download from here:
http://upload.openmv.io/openmv-ide-windows-2.1.0/openmv-ide-windows-2.1.0.exe

closed 2018-09-07 by iabdalkader