openmv/openmv #370: importError: cannot import name CAN

gdramos, 2018-09-01:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Unable to import CAN module on OpenMV Cam M7

### Describe your setup

**IDE version:**
2.0.0
**Firmware version:**
3.0.0
**Operating system:**
Windows 10
### For developers

**GCC version:**

**Build target:**

iabdalkader, 2018-09-02:
from pyb import  CAN

bauer-san, 2018-09-03:
I have the same error with the same system configuration with the import stated by @iabdalkader 
Why is it closed?

iabdalkader, 2018-09-03:
@bauer-san @gdramos CAN was disabled by the last MP update and I re-enabled it in bdacce048dc4fbf3d8afae9b146cd87ffc87b24d

It will be re-enabled in the next release.

>> Why is it closed?

Because it's fixed, I should've mentioned that :) re-opened it if you would like to add something.

gdramos, 2018-09-03:
Oh ok. Cool. I figured it was something like that.
Thanks, @iabdalkader!

bauer-san, 2018-09-03:
Thank you for the info.
Is the next release already planned?  What is your typical release cadence?

iabdalkader, 2018-09-03:
>> Is the next release already planned? What is your typical release cadence?

Yes very soon, typically 1-3 months

iabdalkader, 2018-10-02:
Hi, the new FW is finally out:

https://github.com/openmv/openmv/releases/tag/3.1

closed 2019-05-11 by iabdalkader

x79133000, 2019-06-24:
FW ver. 4.3.1
IDE ver. 2.2.0
Unable to import CAN module on OpenMV Cam H7. Is it disabled again?

iabdalkader, 2019-06-24:
@x79133000 No it's still not supported on the H7 yet, it will be supported in the next release.

x79133000, 2019-06-24:
Thank you for the update.