openmv/openmv #379: USB_VCP.isconnected() not working as expected

phantom-8, 2018-09-20:
### Steps to reproduce
1.  Connect OpenMV Cam M7 to Mac running OpenMV IDE.
2.  The USB.VCP.isconnected() API in the Python program returns True.

For details, refer to the following forum thread as Nyameke asked me to report the bug here.
http://forums.openmv.io/viewtopic.php?f=6&t=936

### Expected behaviour
The USB.VCP.isconnected() should return False when the Cam is connected to IDE.  

If the IDE is not running and VCP port is acting as a serial terminal and communicating with another application on the host computer via the VCP port, the API would return True which is correct in the current implementation.

### Actual behaviour
The USB.VCP.isconnected() currently returns True when the Cam is connected to IDE.  

### Describe your setup

**IDE version:**
OpenMV IDE 2.0.0

**Firmware version:**
OpenMV M7 3.0.0

**Operating system:**
MacOS 10.13.6

iabdalkader, 2018-09-20:
Actually this may not be related to the other issue, let me have a look and get back to you.

iabdalkader, 2018-09-20:
Okay this is not a bug, both the IDE and terminal use VCP (with different baudrates) so technically is_connected is working as expected. I can add a new function for your project vcp.debug_mode_enabled, it will be available in the next release.

PR #380 (2018-09-20): Add function to check if IDE is connected (debug mode).
Fixes #379
closed 2018-09-20 by iabdalkader