openmv/openmv #381: Cascade converter for haar cascade.xml file

gahigg1, 2018-09-20:
When I use cascade_converter.py, the cascade file is created but it is empty and I get this error:

(base) C:\Users\gahig\Desktop\openmv>cascade_convert.py haarcascade_upperbody.xml
Converting new XML format..
Traceback (most recent call last):
  File "C:\Users\gahig\Desktop\openmv\cascade_convert.py", line 422, in <module>
    main()
  File "C:\Users\gahig\Desktop\openmv\cascade_convert.py", line 419, in main
    cascade_binary_universal(args.file, args.stages, args.name)
  File "C:\Users\gahig\Desktop\openmv\cascade_convert.py", line 101, in cascade_binary_universal
    cascade_binary(path, n_stages, name)
  File "C:\Users\gahig\Desktop\openmv\cascade_convert.py", line 154, in cascade_binary
    fout.write(struct.pack('i', size[0]))
TypeError: write() argument must be str, not bytes

Can anyone help? I am using Anaconda prompt on Windows 10.

iabdalkader, 2018-09-20:
Hi, users have reported that the cascade_convert.py script doesn't work on Windows.

gahigg1, 2018-09-20:
Okay, I am getting Lubuntu via virtual box to do it.

iabdalkader, 2018-09-24:
Can you attach the cascade ?

Franzisdrak, 2019-03-04:
Hey there,

i want to add to this topic:
I am running on windows and found that the problem stems from the open cmd with 'w' instead of 'wb'. 

However then i am getting the following error:
Traceback (most recent call last):
  File "convert.py", line 416, in <module>
    main()
  File "convert.py", line 413, in main
    cascade_binary_universal(args.file, args.stages, args.name)
  File "convert.py", line 88, in cascade_binary_universal
    xmldoc = minidom.parse(path)
  File "C:\Users\siege\Anaconda3\lib\xml\dom\minidom.py", line 1958, in parse
    return expatbuilder.parse(file)
  File "C:\Users\siege\Anaconda3\lib\xml\dom\expatbuilder.py", line 911, in parse
    result = builder.parseFile(fp)
  File "C:\Users\siege\Anaconda3\lib\xml\dom\expatbuilder.py", line 211, in parseFile
    parser.Parse("", True)
xml.parsers.expat.ExpatError: no element found: line 1, column 0

Any ideas?

iabdalkader, 2019-03-07:
@Franzisdrak Thank you this fixed the issue on Windows. As for that error, could be the cascade, note this script only supports the most basic Haar Cascade type.

Franzisdrak, 2019-03-12:
Did you try the opencv Haar Cascades? Or are these too complex?

iabdalkader, 2019-03-12:
> Did you try the opencv Haar Cascades? Or are these too complex?

Yes, most the cascades here from open-CV
https://github.com/openmv/openmv/tree/master/ml/haarcascade/cascades

iabdalkader, 2020-07-14:
Cascade converter has been updated a few times, fixed some issues, please reopen if there's still an issue.

closed 2020-07-14 by iabdalkader