openmv/openmv #393: lenet on M7

Franzisdrak, 2018-10-17:
Hey there,

i am using the M7 and i find it very dissapointing, that you show a video running the lenet with the title saying M7/H7, when in reality it is not possible to run on the M7 due to Heap Size. Are there any efforts on making the network even smaller? 

Also i would like to know how far the tensorflow support is?

King regards 
Franzisdrak

iabdalkader, 2018-10-17:
Hi, actually I think @kwagyeman trained a new (smaller) network for the F7, I'll just wait for him to comment.
Re tensorflow, someone is working on that right now, there are still some issues to fix. As soon as it works we'll let you know.

kwagyeman, 2018-10-17:
Yeah, actually, the chars74k net can fit on the M7 if you constrain it to 0-9. This wouldn't be hard to make.

closed 2019-09-09 by iabdalkader