openmv/openmv #411: Requesting messaging over the high speed USB connection

chrisbuerginrogers, 2018-11-12:
Currently I am able to read the frame buffer at will - but any other information transfer is through the "print" statement, which is unidirectional and fills the Tx_buffer unless checked often.  One can also do messaging through SPI / UART / I2C pins.

### Steps to reproduce
1. Try to send something to a running python code on the OpenMV

### Expected behaviour
a few more buffers:
1. a write queue - that can be read by the python code (the opposite of the print statement / Tx_buffer which currently can run like a queue for publishing information)

2. a read tag - like the fb buffer only it contains the latest of some text instead of an image

3. a write tag - the opposite of the fb buffer - something that the PC continually fills and the python code on the OpenMV reads at will 

I think al that is necessary is having these be pre-sized FIFOs (maybe selectable?) and act with identical protocol to the attr calls and reads but are strings.  #1 appends, and #2,3 replaces data every time they are called.  

### Actual behaviour
nothing

### Describe your setup

**IDE version:**
2.0.0

**Firmware version:**
3.0.0

**Operating system:**
MAC and PC

### For developers

**GCC version:**

**Build target:**

Thanks

kwagyeman, 2018-11-12:
Thanks for posting your idea. We'll get on it.

kwagyeman, 2020-05-04:
Hi, we are about to release a new interface library for the OpenMV Cam which provides a standard RPC interface for it. Using this library you are free to build whatever you want the camera to do. Please let us know after using the RPC library if you still need the IDE to provide more features. Using the RPC library you no longer need to depend on the OpenMV Cam debug protocol. You can just roll your own on top of the RPC library which takes care of the heavy lifting of making a reliable transport protocol.

chrisbuerginrogers, 2020-05-04:
awesome.  Looking forward to playing with it.

iabdalkader, 2020-12-01:
The RPC library has been released for a while, is this still needed ?

kwagyeman, 2020-12-01:
Probably, but, I’m not going to focus on this.

chrisbuerginrogers, 2020-12-02:
Hi - thanks all - sorry to be slow - I completely dropped the ball in May (I will chalk it up to being department chair amidst covid - I kinda dropped a lot of things to make sure everyone graduated :).  This RPC stuff is great - assuming both platforms support it.  I will try and get it all working with the LEGO SPIKE Prime...  But yes, this is definitely a working solution.  Thanks

closed 2020-12-02 by chrisbuerginrogers